Genesis-Embodied-AI/genesis-world #252: 我是在一个gpu平台上租的显卡，现在的问题是检测不到显示屏，有什么办法解决吗

wozwdaqian, 2024-12-23:
在优云智算上租的显卡，但是他实际上是给我一个docker容器使用，安装运行示例代码会出现下面的问题

`(Genesis) root@ca23621e2d9c:/workspace/Genesis# python test.py
[Genesis] [10:22:31] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [10:22:31] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [10:22:31] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [10:22:32] [INFO] Running on [NVIDIA GeForce RTX 4090] with backend gs.cuda. Device memory: 23.64 GB.
[Genesis] [10:22:32] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [10:22:33] [INFO] Scene <d8ddbcd> created.
[Genesis] [10:22:33] [INFO] Building scene <d8ddbcd>...
[Genesis] [10:22:33] [INFO] Compiling simulation kernels...
[Genesis] [10:22:33] [INFO] Building visualizer...
Traceback (most recent call last):
  File "/workspace/Genesis/test.py", line 13, in <module>
    scene.build()
  File "/workspace/Genesis/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/workspace/Genesis/genesis/engine/scene.py", line 565, in build
    self._visualizer.build()
  File "/workspace/Genesis/genesis/vis/visualizer.py", line 110, in build
    self._rasterizer.build()
  File "/workspace/Genesis/genesis/vis/rasterizer.py", line 31, in build
    self._renderer = pyrender.OffscreenRenderer(
  File "/workspace/Genesis/genesis/ext/pyrender/offscreen.py", line 29, in __init__
    self._create(pyopengl_platform)
  File "/workspace/Genesis/genesis/ext/pyrender/offscreen.py", line 193, in _create
    self._platform.init_context()
  File "/workspace/Genesis/genesis/ext/pyrender/platforms/egl.py", line 205, in init_context
    assert eglInitialize(self._egl_display, major, minor)
  File "/root/miniconda3/envs/Genesis/lib/python3.9/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
  File "/root/miniconda3/envs/Genesis/lib/python3.9/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.raw.EGL._errors.EGLError: EGLError(
	err = EGL_NOT_INITIALIZED,
	baseOperation = eglInitialize,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f33c80a48c0>,
		c_long(0),
		c_long(0),
	),
	result = 0
)`

下面是我的代码
`
import genesis as gs
gs.init(backend=gs.gpu)

scene = gs.Scene(show_viewer=False)
cam = scene.add_camera(
    res    = (1280, 960),
    pos    = (3.5, 0.0, 2.5),
    lookat = (0, 0, 0.5),
    fov    = 30,
    GUI    = False
)

scene.build()

# render rgb, depth, segmentation mask and normal map
rgb, depth, segmentation, normal = cam.render(depth=True, segmentation=True, normal=True)

# start camera recording. Once this is started, all the rgb images rendered will be recorded internally
cam.start_recording()

import numpy as np
for i in range(120):
    scene.step()

    # change camera position
    cam.set_pose(
        pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
        lookat = (0, 0, 0.5),
    )
    
    cam.render()

# stop recording and save video. If `filename` is not specified, a name will be auto-generated using the caller file name.
cam.stop_recording(save_to_filename='video.mp4', fps=60)
`

下面是运行glxinfo | grep "OpenGL"的输出
`
OpenGL vendor string: Mesa
OpenGL renderer string: llvmpipe (LLVM 15.0.7, 256 bits)
OpenGL version string: 4.5 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL shading language version string: 4.50
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
`

Riften, 2024-12-23:
试一试这个 issue #50 里面给出的两个 docker，一个是源码编译，一个是直接安装。

wozwdaqian, 2024-12-23:
> 试一试这个 issue #50 里面给出的两个 docker，一个是源码编译，一个是直接安装。

不行，我租的平台本身就是用docker的容器给我使用的，我没用办法使用Dockerfile创建容器。不过我会在自己的电脑上尝试，感谢！

wozwdaqian, 2024-12-26:
> 抱歉我们发现远程gpu是会有render的问题，如果有什么进展欢迎分享！（我们现在人手有点捉急。。。）

辛苦啦。

我测试发现gpu占用只有600MB，然后将示例代码的for循环x10，然后经过1分钟之后，黑窗口会显示画面，但是会很卡XD，在显示的画面中输入i要过十来秒才能看到子选项
[image]

下面是我gpu占用情况
[image]

下面是我运行的代码
```
import numpy as np

import genesis as gs

########################## init ##########################
gs.init(backend=gs.gpu)

########################## create a scene ##########################
scene = gs.Scene(
    viewer_options = gs.options.ViewerOptions(
        camera_pos    = (0, -3.5, 2.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 30,
        res           = (960, 640),
        max_FPS       = 60,
    ),
    sim_options = gs.options.SimOptions(
        dt = 0.01,
    ),
    show_viewer = True,
)

########################## entities ##########################
plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.MJCF(
        file  = 'xml/franka_emika_panda/panda.xml',
    ),
)
########################## build ##########################
scene.build()

jnt_names = [
    'joint1',
    'joint2',
    'joint3',
    'joint4',
    'joint5',
    'joint6',
    'joint7',
    'finger_joint1',
    'finger_joint2',
]
dofs_idx = [franka.get_joint(name).dof_idx_local for name in jnt_names]

############ Optional: set control gains ############
# set positional gains
franka.set_dofs_kp(
    kp             = np.array([4500, 4500, 3500, 3500, 2000, 2000, 2000, 100, 100]),
    dofs_idx_local = dofs_idx,
)
# set velocity gains
franka.set_dofs_kv(
    kv             = np.array([450, 450, 350, 350, 200, 200, 200, 10, 10]),
    dofs_idx_local = dofs_idx,
)
# set force range for safety
franka.set_dofs_force_range(
    lower          = np.array([-87, -87, -87, -87, -12, -12, -12, -100, -100]),
    upper          = np.array([ 87,  87,  87,  87,  12,  12,  12,  100,  100]),
    dofs_idx_local = dofs_idx,
)
# Hard reset
for i in range(150):
    if i < 50:
        franka.set_dofs_position(np.array([1, 1, 0, 0, 0, 0, 0, 0.04, 0.04]), dofs_idx)
    elif i < 100:
        franka.set_dofs_position(np.array([-1, 0.8, 1, -2, 1, 0.5, -0.5, 0.04, 0.04]), dofs_idx)
    else:
        franka.set_dofs_position(np.array([0, 0, 0, 0, 0, 0, 0, 0, 0]), dofs_idx)

    scene.step()

# PD control
for i in range(12500):
    if i == 0:
        franka.control_dofs_position(
            np.array([1, 1, 0, 0, 0, 0, 0, 0.04, 0.04]),
            dofs_idx,
        )
    elif i % 250 == 0:
        franka.control_dofs_position(
            np.array([-1, 0.8, 1, -2, 1, 0.5, -0.5, 0.04, 0.04]),
            dofs_idx,
        )
    elif i % 500 == 0:
        franka.control_dofs_position(
            np.array([0, 0, 0, 0, 0, 0, 0, 0, 0]),
            dofs_idx,
        )
    elif i % 750 == 0:
        # control first dof with velocity, and the rest with position
        franka.control_dofs_position(
            np.array([0, 0, 0, 0, 0, 0, 0, 0, 0])[1:],
            dofs_idx[1:],
        )
        franka.control_dofs_velocity(
            np.array([1.0, 0, 0, 0, 0, 0, 0, 0, 0])[:1],
            dofs_idx[:1],
        )
    elif i % 1000 == 0:
        franka.control_dofs_force(
            np.array([0, 0, 0, 0, 0, 0, 0, 0, 0]),
            dofs_idx,
        )
    # This is the control force computed based on the given control command
    # If using force control, it's the same as the given control command
    print('control force:', franka.get_dofs_control_force(dofs_idx))

    # This is the actual force experienced by the dof
    print('internal force:', franka.get_dofs_force(dofs_idx))

    scene.step()
```

lovelyyoshino, 2024-12-26:
你好，你是怎么样在可视化界面中出现弹窗的，我也在优云智算租的显卡，但是无法弹窗

wozwdaqian, 2024-12-26:
> 你好，你是怎么样在可视化界面中出现弹窗的，我也在优云智算租的显卡，但是无法弹窗

我安装了tigervnc

lovelyyoshino, 2024-12-26:
有教程嘛，我装了还得要装xvfb

wozwdaqian, 2024-12-27:
> 有教程嘛，我装了还得要装xvfb

没XD，我问chatgpt，按它的回答来的

lovelyyoshino, 2024-12-27:
就是在docker里面安装了可视化嘛，我在docker里面安装后发现可视化不支持直接弹窗

wozwdaqian, 2024-12-28:
> workspace

我是用tigervnc连接5901端口，然后在连接的窗口中运行的

lovelyyoshino, 2024-12-29:
我也是这么做的安装了sudo apt install ubuntu-desktop和sudo apt-get install tightvncserver。但是出现了可视化需要重新开一个端口才能显示的问题
[image]

duburcqa, 2025-02-28:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/pull/807

closed 2025-02-28 by duburcqa

liguet, 2025-03-28:
> 我也是这么做的安装了sudo apt install ubuntu-desktop和sudo apt-get install tightvncserver。但是出现了可视化需要重新开一个端口才能显示的问题 [image]

你那边也是在原来窗口显示的话会黑屏吗？