Genesis-Embodied-AI/genesis-world #253: Issues in Walking Strategies for Humanoid Robots

formoree, 2024-12-23:
Recently, I have been modifying the demo of a quadruped robot to develop a walking project for a humanoid robot. However, I found that I need to obtain the ground contact forces of the robot while formulating the walking strategy, but it seems that I could not find this in the API. I noticed on the roadmap that the walking strategy is under development, so I would like to inquire whether this API has not been developed yet or if I simply could not locate it.

ziyanx02, 2024-12-23:
Here is an example of getting contact force: 
https://github.com/ziyanx02/Genesis-backflip/blob/main/locomotion_env.py#L432
If you are working on locomotion tasks, that repo could be a good start.

closed 2024-12-23 by ziyanx02

formoree, 2024-12-23:
Thank you very much

NirajPudasaini, 2025-01-13:
@formoree were you able to integrate rsl_rl with humanoid?