Genesis-Embodied-AI/genesis-world #254: solve_cubic is wrong

yliu000, 2024-12-23:
in file genesis/utils/bvh.py

To my understanding, "solve_cubic" should be solving a cubic equation.

But running solve_cubic(1, 0, 0, 1)  (x^3+1),  it returns:
1, -0.000000, 0.000000, 1.000000

This is wrong.

Looking at code:
def solve_cubic(a, b, c, d):
    nr = 0
    x0 = x1 = x2 = 1.0
    ncrit, xc0, xc1 = solve_quadratic(3 * a, 2 * b, c)
    if ncrit == 0:
        x0 = newtons_method(a, b, c, d, xc0, 0)
        nr = 1
    elif ncrit == 1:
        nr, x0, x1 = solve_quadratic(3 * a, 2 * b, c)
    ...
    
    
I don't understand why when "ncrit==1", it returns solve_quadratic(3*a, 2*b, c)

YilingQiao, 2024-12-23:
Hi, @yliu000 We are not using `bvh.py` in this version. How did you get there?

yliu000, 2024-12-24:
OK. I was just browsing the code.
I'll close issue then.

closed 2024-12-24 by yliu000