Genesis-Embodied-AI/genesis-world #255: Hello Genesis got error function 'glGetUniformLocation' not found

zhangiser, 2024-12-23:
I'm on windows used cuda env,run the Hello Genesis,and got this error,i have installed PyOpenGL.
Does my OpenGL env wrong?
[image]
the code：
```python
import genesis as gs
gs.init(backend=gs.gpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()
```
and the error message：
```txt
[Genesis] [11:15:47] [INFO] ╭───────────────────────────────────────────────╮
[Genesis] [11:15:47] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [11:15:47] [INFO] ╰───────────────────────────────────────────────╯
[Genesis] [11:15:49] [INFO] Running on [NVIDIA GeForce RTX 4080 Laptop GPU] with backend gs.cuda. Device memory: 11.99 GB.
[Genesis] [11:15:49] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [11:15:49] [INFO] Scene <0a8197b> created.
[Genesis] [11:15:49] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <760d381>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [11:15:49] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <6795ad6>, morph: <gs.morphs.MJCF(file='D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\genesis\assets\xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [11:15:49] [INFO] Building scene <0a8197b>...
[Genesis] [11:16:00] [INFO] Compiling simulation kernels...
[Genesis] [11:16:14] [INFO] Building visualizer...
Traceback (most recent call last):
  File "D:\code4\genesis\test.py", line 10, in <module>
    scene.build()
  File "D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
  File "D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
  File "D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
  File "D:\Users\zhangiser\anaconda3\envs\torch_env\lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
AttributeError: function 'glGetUniformLocation' not found
[Genesis] [11:16:15] [INFO] 💤 Exiting Genesis and caching compiled kernels...

```

ssm1984119, 2024-12-23:
I met the same problem

AUTUMN9452, 2024-12-23:
I replaced the original platform with Taichi's built-in GUI and visualized particles in real-time.：import torch
import matplotlib.pyplot as plt
import genesis as gs
import taichi as ti

gs.init()
# create a scene
scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt=4e-3,
        substeps=10,
    ),
    sph_options=gs.options.SPHOptions(
        lower_bound=(-0.5, -0.5, 0.0),
        upper_bound=(0.5, 0.5, 1),
        particle_size=0.01,
    ),
    vis_options=gs.options.VisOptions(
        visualize_sph_boundary=True,
    ),
    show_viewer=False,
)

# entities
plane = scene.add_entity(
    morph=gs.morphs.Plane(),
)

liquid = scene.add_entity(
    # viscous liquid
    # material=gs.materials.SPH.Liquid(mu=0.02, gamma=0.02),
    material=gs.materials.SPH.Liquid(),
    morph=gs.morphs.Box(
        pos=(0.0, 0.0, 0.65),
        size=(0.4, 0.4, 0.4),
    ),
    surface=gs.surfaces.Default(
        color=(0.4, 0.8, 1.0),
        vis_mode='particle',
    ),
)

# if i in range(0, 200, 10):
#     draw(i, particles)
scene.build()

window = ti.ui.Window("SPH particles", (768, 768))
canvas = window.get_canvas()
canvas.set_background_color((1, 1, 1))
camera_scene = ti.ui.Scene()
camera = ti.ui.Camera()
camera.position(0.8, 0.1, 0.7)  # 相机 位置坐标
camera.lookat(0, 0.1, 0.1)
camera_scene.set_camera(camera)

position_field = ti.Vector.field(3, dtype=ti.f32, shape=100000)

while window.running:
    horizon = 10
    for i in range(horizon):
        scene.step()
        position = liquid.get_particles()
        position_field.from_numpy(position)

    camera_scene.point_light((1, 1, 1), color=(1, 1, 1))
    camera_scene.particles(position_field, color=(0.4, 0.1, 0.6), radius=0.0025)

    canvas.scene(camera_scene)
    window.show()

fengluying, 2024-12-24:
我也遇到了相同的问题：
AttributeError: function 'glGetUniformLocation' not found

MiangChen, 2024-12-25:
I meet the same bug.
OS: windows 11
python: 3.10
```  File "C:\ProgramData\anaconda3\envs\resnet50\lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
AttributeError: function 'glGetUniformLocation' not found```

zhangiser, 2024-12-25:
finally，i found the problem,because of Interactive Viewer and Headless Rendering is not available on windows,refer [https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html](url)
[image]

gitced, 2024-12-25:
Same problem here on windows.
It works well with scene = gs.Scene(show_viewer=False) but of course i see nothing.
So i guess it's just not possible to get the viewer on windows yet (as indicated by the chart hereabove), which seem to make genesis kind of useless (at least untestable) on windows. :-(
Hope this will be fixed in the future, genesis seems quite fantastic.

ziyanx02, 2024-12-27:
Close as a duplicate of #12.

closed 2024-12-27 by ziyanx02