Genesis-Embodied-AI/genesis-world #259: OpenGL.error.GLError: GLError 	err = 12289

dustin-srp, 2024-12-23:
[Genesis] [04:24:51] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [04:24:51] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [04:24:51] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [04:24:51] [INFO] Running on [Intel(R) Xeon(R) Platinum 8463B] with backend gs.cpu. Device memory: 2015.49 GB.
[Genesis] [04:24:52] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [04:24:52] [INFO] Scene <061abd1> created.
[Genesis] [04:24:52] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <65f4123>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [04:24:52] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <f79b531>, morph: <gs.morphs.MJCF(file='/usr/local/lib/python3.10/dist-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [04:24:52] [INFO] Building scene <061abd1>...
[Genesis] [04:24:56] [INFO] Compiling simulation kernels...
[Genesis] [04:25:02] [INFO] Building visualizer...
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: egl: failed to create dri2 screen
Traceback (most recent call last):
  File "/root/hello_genesis.py", line 29, in <module>
    scene.build()
  File "/usr/local/lib/python3.10/dist-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/genesis/engine/scene.py", line 556, in build
    self._visualizer.build()
  File "/usr/local/lib/python3.10/dist-packages/genesis/vis/visualizer.py", line 110, in build
    self._rasterizer.build()
  File "/usr/local/lib/python3.10/dist-packages/genesis/vis/rasterizer.py", line 31, in build
    self._renderer = pyrender.OffscreenRenderer(
  File "/usr/local/lib/python3.10/dist-packages/genesis/ext/pyrender/offscreen.py", line 29, in __init__
    self._create(pyopengl_platform)
  File "/usr/local/lib/python3.10/dist-packages/genesis/ext/pyrender/offscreen.py", line 193, in _create
    self._platform.init_context()
  File "/usr/local/lib/python3.10/dist-packages/genesis/ext/pyrender/platforms/egl.py", line 205, in init_context
    assert eglInitialize(self._egl_display, major, minor)
  File "/usr/local/lib/python3.10/dist-packages/OpenGL/platform/baseplatform.py", line 402, in __call__
    return self( *args, **named )
  File "/usr/local/lib/python3.10/dist-packages/OpenGL/error.py", line 228, in glCheckError
    raise GLError(
OpenGL.error.GLError: GLError(
	err = 12289,
	baseOperation = eglInitialize,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f7a8c5b5540>,
		c_long(0),
		c_long(0),
	),
	result = 0
)

jiangziben, 2024-12-25:
I have face the sane problem, have you solved it?

soooocold, 2024-12-25:
我遇到了类似的问题
最初的错误
`OpenGL.raw.EGL._errors.EGLError: EGLError(
	err = EGL_NOT_INITIALIZED,
	baseOperation = eglInitialize,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7f190e7fa060>,
		c_long(0),
		c_long(0),
	),
	result = 0
)`
通过重新安装 PyOpenGL
`pip install PyOpenGL==3.1.1a1
`
出现新的问题
`OpenGL.error.GLError: GLError(
	err = 12289,
	baseOperation = eglInitialize,
	cArguments = (
		<OpenGL._opaque.EGLDisplay_pointer object at 0x7facec359eb0>,
		c_long(0),
		c_long(0),
	),
	result = 0
)
`

jiangziben, 2024-12-25:
> I have face the sane problem, have you solved it?

I solved this problem by seting ''export EGL_DEVICE_ID=1''

duburcqa, 2025-02-27:
Unfortunately I do not think this issue is actionable at this point. Can you provide more details about your setup ? Native linux or VM ? Intel CPU or AMD CPU ? ARM or x86 ? Discrete Nvidia GPU ? ...

duburcqa, 2025-02-28:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/pull/807

closed 2025-02-28 by duburcqa

duburcqa, 2025-04-25:
Actually It was only truly fixed recently by #1071