Genesis-Embodied-AI/genesis-world #262: Adding user-defined IK chains

Hardy-cool, 2024-12-23:
For a humanoid robot, when I only want to solve the inverse solution of the arm part, the current solver can only solve the inverse solution of the entire robot link chain, or fix the position outside the arm as the current position. Can this part be optimized to make it more convenient to use?

Genesis-Embodied-AI, 2024-12-24:
Sounds like a very helpful feature! -- if possible, do you think you can help us with this? Implementation should be straightforward, just adding some mask when computing the jacobian udpate. We are really in a shortage of manpower :( If it's not urgent, we will try to add this in the next version bump

Hardy-cool, 2024-12-26:
sure, I'll try the way you said

closed 2025-01-03 by zswang666