Genesis-Embodied-AI/genesis-world #265: TypeError: No array-type handler for type _ctypes.type

Huisouan, 2024-12-23:
I got this error when trying to render the 

[image]

way to reprocude:

[image]

IaMaAVerYRANdOmPerSoN, 2024-12-24:
Same here

ctypes.ArgumentError: ("argument 2: TypeError: No array-type handler for type _ctypes.type (value: <cparam 'P' (0x7f778de18420)>) registered", (1, <cparam 'P' (0x7f778de18420)>))

Pratz1337, 2024-12-25:
Hey, Did you find anything any way to fix this shit

Huisouan, 2024-12-29:
i tried to use Python3.10 and reinstalled the conda env ,and the problem was solved.

Pratz1337, 2024-12-29:
> i tried to use Python3.10 and reinstalled the conda env ,and the problem was solved.

Hi, Thanks for the reply. It worked!

duburcqa, 2025-02-27:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/pull/782

closed 2025-02-27 by duburcqa