Genesis-Embodied-AI/genesis-world #268: ubuntu server 22.04  + 多卡4090 + vnc远程gnome桌面， Building visualizer... 总是出错

zzwu, 2024-12-23:
其它版本信息：

cuda 12.2
drvier 535.54.03

截图：

[image]

日志：

如截图。

另外，show_viewer=False运行时是ok的。

请大佬帮忙分析解决下～🙏

zhouxian, 2024-12-23:
我们注意到远程的设置下是会有一些问题，抱歉！我们会研究下怎么解决。现阶段人手有些不够哈哈

zzwu, 2024-12-23:
> 我们注意到远程的设置下是会有一些问题，抱歉！我们会研究下怎么解决。现阶段人手有些不够哈哈

感谢，期待中～～

Riften, 2024-12-23:
#50 这个问题我遇到过，然后这个 issue 里面我用的 Dockerfile 解决的问题就包含这个。你可以参考一下换 PyOpenGL 版本。

SuanCai-Fish, 2024-12-24:
是的，我这里也是因为使用远程桌面卡住了。发现目前远程服务器（包括租google服务器的）只能用Headless跑，如果使用可视化交互界面就不行。而具有显示器的服务器就没问题。我使用向日葵和VNC连接在同一个具有显示器服务器上测试，同样的配置下，VNC不能使用可视化，界面卡在黑色很久，在20分钟后才有反应。

> 我们注意到远程的设置下是会有一些问题，抱歉！我们会研究下怎么解决。现阶段人手有些不够哈哈

duburcqa, 2025-03-23:
This issue has been fixed on main branch.

closed 2025-03-23 by duburcqa