Genesis-Embodied-AI/genesis-world #275: LuisaRenderPy CUDA backend won't install

kaspar-gestalt, 2024-12-23:
I followed the [instructions ](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#setup)

I have an Ubuntu20.04. 

`NVIDIA-SMI 535.183.01             Driver Version: 535.183.01   CUDA Version: 12.2`

`NVIDIA GeForce GTX 1080 Ti`

```$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Sun_Jul_28_19:07:16_PDT_2019
Cuda compilation tools, release 10.1, V10.1.243
```

wangyian-me, 2024-12-23:
Try coda 12.1 instead of 10.1 perhaps?
Also, I met with other problems before running Luisa on 1080 Ti even with correct cuda and environment.

oscar-lima, 2024-12-25:
I also have Ubuntu 20.04 and I got LuisaRenderPy to acknowledge my installation of CUDA, here is what I did:

I followed steps described in https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#photo-realistic-ray-tracing-rendering

up until:

```
Build LuisaRender. Remember to use the correct cmake.
```

meaning, do NOT try:
```bash
cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON # remember to check python version
```
because you want to carefully configure CUDA first. If you do, the following warnings appear:

# LuisaRenderPy cmake CUDA warnings

During cmake configuration of LuisaRenderPy the following warning appeared:
```
foo@bar $ cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.10 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF -DCMAKE_INSTALL_PREFIX=/home/oscar/.pyenv/versions/genesis-sim/lib/python3.10/site-packages
.
.
CMake Warning at src/compute/scripts/validate_options.cmake:70 (message):
  The CUDA backend is not available.  The CUDA backend will be disabled.
Call Stack (most recent call first):
  src/compute/scripts/validate_options.cmake:98 (report_feature_not_available)
  src/compute/CMakeLists.txt:53 (include)
.
.
.
```

However LuisaRenderPy compiles despite the warnings above, at this stage if you try the rendering [demo.py](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/rendering/demo.py) you get a Runtime error and the program crashes:
```
foo@bar $ python demo.py --vis
.
.
[2024-12-24 08:32:00.444] [console] [error] Backend 'cuda' is not installed. [... Genesis/genesis/ext/LuisaRender/src/compute/src/runtime/context.cpp:77]
.
.
.
Aborted (core dumped)
```

So LuisaRenderPy really needs a clean, solid CUDA setup to work. Based on @wangyian-me's comment, I gave CUDA 12.1 a try. However, from #207, I see that people are trying CUDA 12.2 with success (for Ubuntu 22.04).

# Check the maximum CUDA version supported by the installed NVIDIA driver

One thing one needs to be careful about is that when doing ```nvidia-smi``` the CUDA version that is printed is the maximum CUDA version supported by the installed NVIDIA driver and NOT your "active" CUDA version, e.g.,:
```bash
foo@bar $ nvidia-smi
Tue Dec 24 00:05:39 2024
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.183.01             Driver Version: 535.183.01   CUDA Version: 12.2(A)  |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  Quadro RTX 4000 with Max...    Off | 00000000:01:00.0  On |                  N/A |
| N/A   48C    P8               8W /  30W |    853MiB /  8192MiB |      3%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
[... 117 lines cut ...]
Finally CUDA was found!

```
-- Found CUDAToolkit: /usr/local/cuda-12.1/targets/x86_64-linux/include (found suitable version "12.1.66", minimum required is "11.7")
-- Build with CUDA backend: 12.1.66
-- Downloading nvcomp from https://developer.download.nvidia.com/compute/nvcomp/3.0.4/local_installers/nvcomp_3.0.4_x86_64_12.x.tgz
-- Populating nvcomp
```

Now, to be honest... I still have vulkan and other issues (see F), however since this issue is concerned with "LuisaRenderPy CUDA backend won't install" (for ubuntu 20.04), I think I covered at least that part...

F:
```
Warnings:
No package 'zzip-zlib-config' found # SOLVED via: sudo apt install libzzip-dev
The DirectX backend is not available.  The DirectX backend will be disabled.
The Metal backend is not available.  The Metal backend will be disabled.
Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
The Vulkan backend is not available.  The Vulkan backend will be disabled.
Could not find sccache
NVTT not found! Please install NVTT from
  https://developer.nvidia.com/nvidia-texture-tools-exporter and set the
  CMake NVTT_DIR variable to the folder containing nvtt*.dll (e.g.
  C:\Program Files\NVIDIA Corporation\NVIDIA Texture Tools) .
```

kaspar-gestalt, 2024-12-26:
Cool, thanks @oscar-lima @wangyian-me .

I installed CUDA toolkit 12.6 and CUDA drivers 565.57.01, and re-built LuisaRenderPy. Now I am encountering the error `[error] CUDA_ERROR_INVALID_SOURCE: device kernel image is invalid [/home/kaspar/repos/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda
/cuda_shader_native.cpp:46]`, when I try to `cam.render(...)` 

Is that directly related, or should I open a new issue?

kaspar-gestalt, 2025-01-03:
@oscar-lima Does the actual compilation of LuisaRender work now with NVTT enabled, or just its build configuration? I'm asking because my configuration also seems find, with NVTT being found, but the compilation errors out: https://github.com/Genesis-Embodied-AI/Genesis/issues/465

kaspar-gestalt, 2025-01-10:
This issue was solved by installing a different combination of toolkit/driver versions. https://github.com/Genesis-Embodied-AI/Genesis/issues/465