Genesis-Embodied-AI/genesis-world #279: Assign an initial velocity to the entity.

Xujxyang, 2024-12-24:
Thank you for the excellent work! I would like to use the MPM solver organized here to perform some simulations. However, I noticed that it doesn't seem possible to directly set the initial velocity for an entity in add_entity() https://github.com/Genesis-Embodied-AI/Genesis/blob/da7c36319dcdc8d235f2ba8ed81baf17b98127ff/genesis/engine/scene.py#L220 or gs.morphs https://github.com/Genesis-Embodied-AI/Genesis/blob/da7c36319dcdc8d235f2ba8ed81baf17b98127ff/genesis/options/morphs.py#L20 
How should I achieve this?

zhenjia-xu, 2024-12-24:
You can manually set velocity after the scene is built.
Please use this function: [link](https://github.com/Genesis-Embodied-AI/Genesis/blob/257ca2cb186bf17b720be3f93eee803668ca596d/genesis/engine/entities/particle_entity.py#L296)
Here is an example: [link](https://github.com/Genesis-Embodied-AI/Genesis/blob/257ca2cb186bf17b720be3f93eee803668ca596d/examples/differentiable_push.py#L102)

Xujxyang, 2024-12-24:
Thank you very much for your response! However, I’ve encountered another tricky issue.

I am running a program on a server with a single NVIDIA 4090 GPU. When executing this line of code https://github.com/Genesis-Embodied-AI/Genesis/blob/257ca2cb186bf17b720be3f93eee803668ca596d/examples/differentiable_push.py#L89 , the method here https://github.com/Genesis-Embodied-AI/Genesis/blob/00d041d1bd0306c3e3c44cc9f2c70d3a0448b55e/genesis/ext/pyrender/platforms/egl.py#L57 returns a count of 0, which causes this section  https://github.com/Genesis-Embodied-AI/Genesis/blob/00d041d1bd0306c3e3c44cc9f2c70d3a0448b55e/genesis/ext/pyrender/platforms/egl.py#L82  to fail in retrieving the GPU ID. As a result, the program cannot proceed further. Do you have any suggestions or solutions for this issue?

YilingQiao, 2025-03-06:
The latest version should resolve this problem. Please let us know if the issue persists.

closed 2025-03-06 by YilingQiao