Genesis-Embodied-AI/genesis-world #280: TypeError about tensor

koukoucoconut, 2024-12-24:
Does anyone face it？I have no idea about how to set my device on GPU. Actually, I change the CPU with GPU in the code, but always return to the error.
[image]

zhenjia-xu, 2024-12-24:
Could you please provide the code so that we can debug it?

koukoucoconut, 2024-12-24:
> Could you please provide the code so that we can debug it?

import genesis as gs
gs.init(backend = gs.gpu)
scene = gs.Scene(
    show_viewer = True,
    viewer_options = gs.options.ViewerOptions(
        res           = (1280, 960),
        camera_pos    = (3.5, 0.0, 2.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 40,
        max_FPS       = 60,
    ),
    vis_options = gs.options.VisOptions(
        show_world_frame = True,
        world_frame_size = 1.0,
        show_link_frame  = False,
        show_cameras     = False,
        plane_reflection = True,
        ambient_light    = (0.1, 0.1, 0.1),
    ),
    renderer=gs.renderers.Rasterizer(),
)
plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)
cam = scene.add_camera(
    res    = (640, 480),
    pos    = (3.5, 0.0, 2.5),
    lookat = (0, 0, 0.5),
    fov    = 30,
    GUI    = False,
)
scene.build()
cam.start_recording()
import numpy as np
for i in range(120):
    scene.step()
    cam.set_pose(
        pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
        lookat = (0, 0, 0.5),
    )
    cam.render()
cam.stop_recording(save_to_filename='video.mp4', fps=60)

zhenjia-xu, 2024-12-24:
The code you provided can be run successfully on my side. I tested both CPU and GPU backend.
Could you please double-check your pytorch installation? Make sure you installed gpu-version, instead of cpu-version.

koukoucoconut, 2024-12-24:
> The code you provided can be run successfully on my side. I tested both CPU and GPU backend. Could you please double-check your pytorch installation? Make sure you installed gpu-version, instead of cpu-version.
I check my pytorch：
python -c "import torch; print(torch.__version__)"
1.11.0+cu115
confused        Is this version not available？

zhenjia-xu, 2024-12-24:
emmm. it is indeed a cuda version, although it is pretty old.
How about you create a new Python environment and just run 
```
pip install genesis-world
pip install torch
python test1.py
```
this will install the latest pytorch and make sure the python environment is clean enough.

koukoucoconut, 2024-12-25:
> emmm. it is indeed a cuda version, although it is pretty old. How about you create a new Python environment and just run
> 
> ```
> pip install genesis-world
> pip install torch
> python test1.py
> ```
> 
> this will install the latest pytorch and make sure the python environment is clean enough.

thank you. I update my version, it's ok

closed 2025-02-23 by YilingQiao