Genesis-Embodied-AI/genesis-world #281: Invalid device ID (0)

Wykay, 2024-12-24:
Hi,

I am trying to run a demo on a headless RTX3090 server and I meet this error when building the Visualizer. The log is as below
`[Genesis] [06:00:53] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [06:00:53] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [06:00:53] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [06:00:54] [INFO] Running on [NVIDIA GeForce RTX 3090] with backend gs.cuda. Device memory: 23.69 GB.
[Genesis] [06:00:55] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [06:00:56] [INFO] Scene <5be2533> created.
[Genesis] [06:00:56] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <6bd4515>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [06:00:56] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <e592935>, morph: <gs.morphs.MJCF(file='/opt/data/private/jwq/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [06:00:57] [INFO] Building scene <5be2533>...
[Genesis] [06:01:03] [INFO] Compiling simulation kernels...
[Genesis] [06:01:10] [INFO] Building visualizer...
Traceback (most recent call last):
  File "/opt/data/private/jwq/Genesis/demo.py", line 44, in <module>
    scene.build()
  File "/opt/data/private/jwq/Genesis/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/opt/data/private/jwq/Genesis/genesis/engine/scene.py", line 565, in build
    self._visualizer.build()
  File "/opt/data/private/jwq/Genesis/genesis/vis/visualizer.py", line 110, in build
    self._rasterizer.build()
  File "/opt/data/private/jwq/Genesis/genesis/vis/rasterizer.py", line 31, in build
    self._renderer = pyrender.OffscreenRenderer(
  File "/opt/data/private/jwq/Genesis/genesis/ext/pyrender/offscreen.py", line 29, in __init__
    self._create(pyopengl_platform)
  File "/opt/data/private/jwq/Genesis/genesis/ext/pyrender/offscreen.py", line 185, in _create
    egl_device = egl.get_device_by_index(device_id)
  File "/opt/data/private/jwq/Genesis/genesis/ext/pyrender/platforms/egl.py", line 84, in get_device_by_index
    raise ValueError("Invalid device ID ({})".format(device_id, len(devices)))
ValueError: Invalid device ID (0)
[Genesis] [06:01:11] [INFO] 💤 Exiting Genesis and caching compiled kernels...`

and my code is
`import genesis as gs
from genesis.constants import backend as gs_backend

gs.init(backend=gs_backend.gpu) 

scene = gs.Scene(
    show_viewer=False, # Disable viewer since we're recording
)

plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

cam = scene.add_camera(
res=(320, 240), 
pos=(3.5, 0.0, 2.5),
lookat=(0, 0, 0.5),
fov=30,
GUI=False
)

scene.build()

cam.start_recording()

for i in range(5): # Reduced from 1000 to 100 frames
    scene.step()
    # Render only every 2nd frame to improve performance
    if i % 2 == 0:
        cam.render()

cam.stop_recording(save_to_filename='simulation.mp4', fps=30)`

Pratz1337, 2024-12-26:
Did you find a fix/?

Seas00n, 2024-12-31:
When `show_viewer=True` for me, the problem can be solved.

Dragon2938734, 2025-01-09:
hi, i meet the same problem, do you find a solution?

Dragon2938734, 2025-01-09:
> When `show_viewer=True` for me, the problem can be solved.
> 
> It seems that the scene can't build visualizer correctly

but set `show_viewer=True` will meet another bug "[Genesis] [21:31:16] [ERROR] No display detected. Use `show_viewer=False` for headless mode." since i use a server not a local computer.

ZeyuPing, 2025-01-28:
> but set `show_viewer=True` will meet another bug "[Genesis] [21:31:16] [ERROR] No display detected. Use `show_viewer=False` for headless mode." since i use a server not a local computer.

The same problem. Does genesis actually support headless mode?