Genesis-Embodied-AI/genesis-world #282: Hello Genesis missing module Imath

jremington, 2024-12-24:
Error message:
  File "/home/jim/Desktop/Genesis-main/genesis/options/textures.py", line 4, in <module>
    import Imath
ModuleNotFoundError: No module named 'Imath'

I've installed the package in various ways, including this way, but the error persists. What do I need to change to import the installed module?

`jim@jim-12core:~/Desktop/Genesis-main$ pip3 install Imath
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: Imath in /home/jim/.local/lib/python3.10/site-packages (0.0.2)
`

for example:
`jim@jim-12core:~/Desktop/Genesis-main$ python3 test1.py
Traceback (most recent call last):
  File "/home/jim/Desktop/Genesis-main/test1.py", line 1, in <module>
    import genesis as gs
  File "/home/jim/Desktop/Genesis-main/genesis/__init__.py", line 266, in <module>
    from .options import renderers
  File "/home/jim/Desktop/Genesis-main/genesis/options/renderers.py", line 6, in <module>
    from .surfaces import Surface
  File "/home/jim/Desktop/Genesis-main/genesis/options/surfaces.py", line 9, in <module>
    from .textures import ColorTexture, ImageTexture, Texture
  File "/home/jim/Desktop/Genesis-main/genesis/options/textures.py", line 4, in <module>
    import Imath
ModuleNotFoundError: No module named 'Imath'
`

`

zhenjia-xu, 2024-12-24:
Please install OpenEXR
```
pip install OpenEXR
```
BTW, if you run `pip install -e .` it will automatically install all packages needed. Please let us know if it doesn't work for you.

jremington, 2024-12-24:
That got me a little farther! Now missing "coacd"

The pip install -e  fails because the normal site packages are not writeable.

`Successfully installed openEXR-3.3.2
jim@jim-12core:~/Desktop/Genesis-main$ python3 test1.py
Traceback (most recent call last):
  File "/home/jim/Desktop/Genesis-main/test1.py", line 1, in <module>
    import genesis as gs
  File "/home/jim/Desktop/Genesis-main/genesis/__init__.py", line 274, in <module>
    from .engine.scene import Scene
  File "/home/jim/Desktop/Genesis-main/genesis/engine/scene.py", line 5, in <module>
    from genesis.engine.entities import Emitter
  File "/home/jim/Desktop/Genesis-main/genesis/engine/entities/__init__.py", line 1, in <module>
    from .avatar_entity import AvatarEntity
  File "/home/jim/Desktop/Genesis-main/genesis/engine/entities/avatar_entity/__init__.py", line 1, in <module>
    from .avatar_entity import AvatarEntity
  File "/home/jim/Desktop/Genesis-main/genesis/engine/entities/avatar_entity/avatar_entity.py", line 3, in <module>
    from ..rigid_entity import RigidEntity
  File "/home/jim/Desktop/Genesis-main/genesis/engine/entities/rigid_entity/__init__.py", line 1, in <module>
    from .rigid_entity import RigidEntity
  File "/home/jim/Desktop/Genesis-main/genesis/engine/entities/rigid_entity/rigid_entity.py", line 8, in <module>
    from genesis.utils import mesh as mu
  File "/home/jim/Desktop/Genesis-main/genesis/utils/mesh.py", line 7, in <module>
    import coacd
ModuleNotFoundError: No module named 'coacd'
ji`

zhenjia-xu, 2024-12-24:
Are you using the system's default Python?
If so, I highly recommend using `conda` or `venv` so that you are much more flexible in installing packages.

jremington, 2024-12-24:
I haven't used either.

zhenjia-xu, 2024-12-24:
The right way to install igl should be
```
pip install libigl
```

jremington, 2024-12-24:
I see what you mean, thanks.

closed 2024-12-24 by jremington