Genesis-Embodied-AI/genesis-world #283: How to add entity from franka_sim?

milong26, 2024-12-24:
I'm working on ubuntu,my xml folder in `anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/assets/xml/`. I saw there is a folder named `franka_sim` besides `franka_emika_panda` so i want to use it.

When I use
```python
franka = scene.add_entity(
    gs.morphs.MJCF(
        file='xml/franka_sim/bi-franka_panda.xml'
    ),
)
```
to replace code from [hello genesis](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html),it said:
> ValueError: Error: Error opening file '/home/zhouxian/git/Genesis-dev/genesis/assets/xml/franka_sim/meshes/collision/link1.stl': No such file or directory

How to solve this?

zhenjia-xu, 2024-12-24:
Thanks for pointing it out.
This bug is fixed. Please pull the latest repo and give it a try.

milong26, 2024-12-24:
> Thanks for pointing it out. This bug is fixed. Please pull the latest repo and give it a try.

I used pip to install genesis,should I use git clone?

milong26, 2024-12-24:
How to close it

closed 2024-12-24 by zhenjia-xu