Genesis-Embodied-AI/genesis-world #285: libGL error: No matching fbConfigs or visuals found with vscode remote-ssh and ForwardX11

0xdezoe, 2024-12-24:
Hello, guys.

I rented a 4090 server with Ubuntu Server 22.04, and I’m using VSCode’s remote-ssh to run Genesis. I’ve set “ForwardX11 yes” to forward the display, but I encountered an error during runtime.

My environment:
- OS: Ubuntu Server 22.04
- GPU: Nvidia RTX 4090
- Driver Version: 535
- CUDA Version: 12.2
- Python 3.10.12

My ssh config：

> Host 4090-hostname
>     Hostname x.x.x.x
>     Port 22
>     ForwardX11 yes
>     ForwardX11Trusted yes

First of all, I ran a simple UI application like xeyes, and it worked fine, which confirms that my SSH ForwardX11 configuration is correct.

However, when I try to run Genesis (e.g., python examples/tutorials/hello_genesis.py), I encounter the following error:

> libGL error: No matching fbConfigs or visuals found
> libGL error: failed to load driver: swrast
> Exception in thread Thread-2 (_init_and_start_app):
> Traceback (most recent call last):
>   File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.10/threading.py", line 953, in run
>     self._target(*self._args, **self._kwargs)
>   File "/usr/local/lib/python3.10/dist-packages/genesis/ext/pyrender/viewer.py", line 1126, in _init_and_start_app
>     super(Viewer, self).__init__(
>   File "/usr/local/lib/python3.10/dist-packages/pyglet/window/xlib/__init__.py", line 168, in __init__
>     super().__init__(*args, **kwargs)
>   File "/usr/local/lib/python3.10/dist-packages/pyglet/window/__init__.py", line 527, in __init__
>     context = config.create_context(gl.current_context)
>   File "/usr/local/lib/python3.10/dist-packages/pyglet/gl/xlib.py", line 114, in create_context
>     return XlibContext(self, share)
>   File "/usr/local/lib/python3.10/dist-packages/pyglet/gl/xlib.py", line 149, in __init__
>     raise gl.ContextException(msg)
> pyglet.gl.ContextException: Could not create GL context

[image]

How can I resolve this?

Colmar-zlicheng, 2024-12-24:
Same issue.

Colmar-zlicheng, 2024-12-24:
I have same issue using vncserver.

duburcqa, 2025-03-22:
This kind of issues has been fixed on main branch.

closed 2025-03-22 by duburcqa