Genesis-Embodied-AI/genesis-world #287: which kinds of mjcf file it support? I used xml from mujoco/models but

milong26, 2024-12-24:
I used `arm26.xml` from [mujoco/models](https://github.com/google-deepmind/mujoco/blob/main/model/tendon_arm/arm26.xml). It shows well in mujoco.

[image]

But when I try to use it in genesis,it only shows a simple plane,lack the entity i added.

My whole code about entity is:
```python
plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.MJCF(
        file='robot/arm26.xml',
    ),
)

```

[image]

genesis version latest.

Kashu7100, 2024-12-24:
I will take a look

lupinjia, 2024-12-24:
same problem, can't visualize it

milong26, 2024-12-25:
xD I tested many exmaples and find somewhat may casue this

contype="0" conaffinity="0"  in xml file in <geom> label.

Wish this may help u

Kashu7100, 2024-12-26:
For MJCF, we currently support following types (PR for adding more types are welcome: check `genesis/utils/mjcf.py`):
- `mjGEOM_PLANE`
- `mjGEOM_SPHERE`
- `mjGEOM_CAPSULE`
- `mjGEOM_CYLINDER`
- `mjGEOM_MESH`

I think `contype` and `conaffinity` decides the mesh type of geom (line 238 of `genesis/utils/mjcf.py`) as:
 
```python
is_col = bool(mj.geom_conaffinity[i_g] or mj.geom_contype[i_g])
```

milong26, 2024-12-26:
> For MJCF, we currently support following types (PR for adding more types are welcome: check `genesis/utils/mjcf.py`):
>
> * `mjGEOM_PLANE`
> * `mjGEOM_SPHERE`
> * `mjGEOM_CAPSULE`
> * `mjGEOM_CYLINDER`
> * [...]

k Since only the 5 types are supported, does it mean we cannot use our own models now? The only way is to construct models we want using these 5

lupinjia, 2024-12-26:
> k Since only the 5 types are supported, does it mean we cannot use our own models now? The only way is to construct models we want using these 5

I think most of the custom models use mesh as geometry(.STL, .dae, etc), so it's viable to use customs models theoretically. However, according to my experience, it's more easy to use URDF than xml description.

mochan-b, 2025-01-07:
If the type of `body` is mesh, then only contype=0 and coaffinity=0 can be visualized. Meaning that meshes are only for visualization and not for collision.

In mujoco, meshes would be convexified and then could collision objects. In mujoco now, mesh can be sdf. Is this implemented here? 

In the UR5 example that works, (https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/assets/xml/universal_robots_ur5e/ur5e.xml), all the meshes are non-collision and collision objects are only capsules. 

Is this correct? 
If we want to make collision objects, they cannot be meshes (didn't check which ones can be collision objects from your list of plane, sphere, capsule, cylinder ). 
If we want to visualize a mesh, they cannot be collision object.

zhouxian, 2025-01-07:
> If the type of `body` is mesh, then only contype=0 and coaffinity=0 can be visualized. Meaning that meshes are only for visualization and not for collision.
>
> In mujoco, meshes would be convexified [...]

Oh no, mesh collision is definite supported. The official tutorial uses Franka mjcf which contains a lot of meshes

mochan-b, 2025-01-07:
> Oh no, mesh collision is definite supported. The official tutorial uses Franka mjcf which contains a lot of meshes

Thank you. That was very insightful. 

I checked the franka and there is a visual stl file and collision stl file. The collision stl files are probably convexified versions of the visual stl files.
For example,
Visual: https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/assets/xml/franka_sim/meshes/visual/finger.stl
Collision: https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/assets/xml/franka_sim/meshes/collision/finger.stl

In my case, I just doubled the `<geom` and made one visual and one collision. That looks like it is doing collision and visuals look correct. 

Better idea probably is to make convex polygons for collision or model it as capsules for my model.

alexanderswerdlow, 2025-02-14:
There should be a warning/exception for non-supported types to prevent this issue.

YilingQiao, 2025-02-15:
Hi @milong26 , we support those MJCF geom types. They’re just not visualized by default because we only render visual meshes, not collision meshes. I didn’t realize this had caused so much confusion. Thanks, @alexanderswerdlow , for bringing this to my attention. We will probably change this behaviour.

To visualize collision meshes, you can do things like
```
    plane = scene.add_entity(
        gs.morphs.MJCF(
            file="./arm26.xml",
            pos=(0, 0, 0),
        ),
        vis_mode="collision",
    )

```

closed 2025-02-23 by YilingQiao