Genesis-Embodied-AI/genesis-world #291: How to control suction in Genesis

yyxssm, 2024-12-24:
Hi, thinks for your great work!

I have want to simulate suction in Genesis and I find that there is a suction model in `genesis/assets/urdf/panda_bullet/panda_suction.urdf`, thanks for your great work again.

I have load the suction model into simulator by using 
```python
import genesis as gs

gs.init(backend=gs.cpu)

scene = gs.Scene(
    show_viewer=True,
)

plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.URDF(
        file='urdf/panda_bullet/panda.urdf',
        fixed=True,
    ),
)

scene.build()
for i in range(1000):
    scene.step()

```

Up to now I have not find how to control the suction to 'grasp' objects. I have found that there is a tutorial in `examples/tutorials/IK_motion_planning_grasp.py`, which is a gripper to grasp object, but I want to suck up an object with a suction and I don't know the exact command.

Could anyone help me? Thank you very much!

zhouxian, 2024-12-24:
Hi, thank you for your interest! Suction requires adding a "attach" constraint between the cup and the object, which we do not support it but will add quickly. For now a quick workaround would be that you manually control the object by `control_dofs_position` to follow the end-effector

yyxssm, 2024-12-24:
Thank you very much!

yyxssm, 2025-01-04:
Hi, thanks for your great work, about how to use the suction cup, I have work on few weeks but still have two questions.

1. **When using the suction cup urdf model in `genesis/assets/urdf/panda_bullet/panda_suction.urdf`，I cannot get the `hand` end effector.** 
    
   There is `panda_leftfinger` link and `panda_rightfinger` link in the urdf, but I cannot find them by using code link `robot.get_link('panda_rightfinger')`.

   I try with the following code

```python
import numpy as np
import genesis as gs

#init
gs.init(backend=gs.gpu)

#create a scene
scene = gs.Scene(
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(3,-1,1.5),
        camera_lookat=(0.0,0.0,0.5),
        camera_fov=30,
        max_FPS=60,
    ),
    sim_options=gs.options.SimOptions(
        dt=0.01,
        substeps=4,
    ),
    show_viewer=True,
)

#load objects
plane = scene.add_entity(
    gs.morphs.Plane(),
)

robot = scene.add_entity(
    gs.morphs.URDF(file='urdf/panda_bullet/panda_suction.urdf', fixed = True),

)

print(robot._links)
```

and the output is belowed, there is no `panda_leftfinger` link and `panda_rightfinger` link, how can I get the endeffector?

```
─────────────── <gs.List> of <gs.RigidLink> ───────────────
<gs.List>(len=8, [
    <gs.RigidLink>: <726b33d>, name: 'panda_link0', idx: 3,
    <gs.RigidLink>: <6a96aea>, name: 'panda_link1', idx: 4,
    <gs.RigidLink>: <3b5a7e9>, name: 'panda_link2', idx: 5,
    <gs.RigidLink>: <b9e179c>, name: 'panda_link3', idx: 6,
    <gs.RigidLink>: <faafa27>, name: 'panda_link4', idx: 7,
    <gs.RigidLink>: <e492ebc>, name: 'panda_link5', idx: 8,
    <gs.RigidLink>: <0b5f796>, name: 'panda_link6', idx: 9,
    <gs.RigidLink>: <17c2610>, name: 'panda_link7', idx: 10,
])
```

zhouxian, 2025-01-04:
We will merge links connected using fixed joint for performance optimization. There’s a parameter in morph.urdf where you can turn this off or add ‘links_to_keep’

yyxssm, 2025-01-04:
This is my second question, I use the gripper as the manipulator and have tried with `control_dofs_position` to control the object to the manipulator, this is my code and effect
```python
import numpy as np
import genesis as gs

# Initialize the simulator with GPU backend
gs.init(backend=gs.gpu)

# Create a scene
scene = gs.Scene(
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(3, -1, 1.5),
        camera_lookat=(0.0, 0.0, 0.5),
        camera_fov=30,
        max_FPS=60,
    ),
    sim_options=gs.options.SimOptions(
        dt=0.01,
        substeps=4,
    ),
    show_viewer=True,
)

# Load objects into the scene
plane = scene.add_entity(
    gs.morphs.Plane(),
)
cube = scene.add_entity(
    gs.morphs.Box(
        size=(0.03, 0.03, 0.03),
        pos=(0.5, 0.5, 0.015),
    )
)

cube_2 = scene.add_entity(
    gs.morphs.Box(
        size=(0.03, 0.03, 0.03),
        pos=(0.2, 0.5, 0.015),
    )
)

# Add robot entity (Panda robot from MJCF)
robot = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

# Set the degrees of freedom for the robot (7 DOF for the body, 2 DOF for the fingers)
motors_dof = np.arange(7)
fingers_dof = np.arange(7, 9)

# Set the initial pose
qpos = np.array([-1.0124, 1.5559, 1.3662, -1.6878, -1.5799, 1.7757, 1.4602, 0.04, 0.04])
robot.set_qpos(qpos)
scene.step()

# Get the end effector (hand)
end_effector = robot.get_link('hand')

# Perform inverse kinematics to reach the target position
qpos = robot.inverse_kinematics(
    link=end_effector,
    pos=np.array([0.5, 0.5, 0.3])
)

# Reach the target position (keep it stable for a while)
for i in range(200):
    print("reach", i)
    robot.control_dofs_position(qpos[:-2], motors_dof)  # Control the robot's body movement
    scene.step()

# Move the robot down to interact with the cube
qpos_down = robot.inverse_kinematics(
    link=end_effector,
    pos=np.array([0.5, 0.5, (cube.get_pos()[2]).cpu().numpy() + 0.1])
)

qpos_back = robot.inverse_kinematics(
    link=end_effector,
    pos=np.array([0.5, -0.1, 0.3])
)

robot.control_dofs_position(qpos_down[:-2], motors_dof)  # Control the robot's body movement

# Move the robot down and maintain stability
for i in range(200):
    print("down", i)
    scene.step()

# Simulate the grasping and movement of the cube
for i in range(200):
    print("suck", i)
    if end_effector.get_pos()[2] <= cube.get_pos()[2] + 0.1:
        robot.control_dofs_position(qpos_back[:-2], motors_dof)
        cube.control_dofs_position([0.5, -0.1, 0.25, 0, 0, 0])
    scene.step()

# Drop the cube onto the ground
for i in range(200):
    cube.control_dofs_position([0.5, -0.1, 0.015, 0, 0, 0])  # Drop position
    scene.step()

# Move the robot to pick up the second cube
qpos_2 = robot.inverse_kinematics(
    link=end_effector,
    pos=np.array([0.2, 0.5, 0.3])
)

for i in range(200):
    robot.control_dofs_position(qpos_2[:-2], motors_dof)
    scene.step()

# Move the robot down to interact with the second cube
qpos_down = robot.inverse_kinematics(
    link=end_effector,
    pos=np.array([0.2, 0.5, (cube_2.get_pos()[2]).cpu().numpy() + 0.1])
)

for i in range(200):
    robot.control_dofs_position(qpos_down[:-2], motors_dof)
    scene.step()

# Simulate the grasping and movement of the second cube
for i in range(200):
    print("suck", i)
    if end_effector.get_pos()[2] <= cube_2.get_pos()[2] + 0.1:
        robot.control_dofs_position(qpos_back[:-2], motors_dof)
        cube_2.control_dofs_position([0.5, -0.1, 0.25, 0, 0, 0])
    scene.step()

# Drop the second cube onto the ground
for i in range(200):
    cube_2.control_dofs_position([0.5, -0.1, 0.015, 0, 0, 0])  # Drop position
    scene.step()

```

This is the effect

https://github.com/user-attachments/assets/94051c08-1a21-4cba-9376-ed4ccacafd15

I noticed that the movement speed of the object and the robot arm is not consistent. Is this a normal phenomenon?  Also, may I ask when the grasping constraints can be added in Genesis?

yyxssm, 2025-01-04:
> We will merge links connected using fixed joint for performance optimization. There’s a parameter in morph.urdf where you can turn this off or add ‘links_to_keep’

Thank you very much, this issue has been resolved!

closed 2025-02-23 by YilingQiao

Xu-Xuanhui, 2025-03-05:
> Hi, thank you for your interest! Suction requires adding a "attach" constraint between the cup and the object, which we do not support it but will add quickly. For now a quick workaround would be that you manually control the object by `control_dofs_position` to follow the end-effector

Hello, may I ask if Genesis now supports  a "attach" constraint between the cup and the object?