Genesis-Embodied-AI/genesis-world #293: error when running hello_genesis.py

zhjc2014, 2024-12-24:
> (genesis) zhan@zhan-robot:~/src/Genesis$  /usr/bin/env /home/zhan/anaconda3/envs/genesis/bin/python /home/zhan/.vscode/extensions/ms-python.debugpy-2024.14.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher 35029 -- /home/zhan/src/Genesis/examples/tutorials/hello_genesis.py 
free(): invalid pointer

zhjc2014, 2024-12-24:
what't the problem

milong26, 2024-12-25:
Can you import genesis well? I mean do like this:
```
(genesis) python
(genesis) import genesis
(genesis) print(genesis.__path__)
```
Im new to this project too,maybe this will help

zhjc2014, 2024-12-25:
> Can you import genesis well? I mean do like this:
> 
> ```
> (genesis) python
> (genesis) import genesis
> (genesis) print(genesis.__path__)
> ```
> 
> Im new to this project too,maybe this will help

the error happen when execute the import line, maybe the pytorch version not compatible?

milong26, 2024-12-25:
> the error happen when execute the import line, maybe the pytorch version not compatible?

Maybe? I think you need to reinstall genesis and pytorch. By the way you can install from github while they're still updating github xD. 

You can `git clone https://github.com/Genesis-Embodied-AI/Genesis.git` then set PYTHONPATH `export PYTHONPATH=/the path where you clone the repo/Genesis`

zhjc2014, 2024-12-27:
> Maybe? I think you need to reinstall genesis and pytorch. By the way you can install from github while they're still updating github xD.
>
> You can `git clone [...]

`(genesis) zhan@zhan-robot:~/src/Genesis$ /home/zhan/anaconda3/envs/genesis/bin/python /home/zhan/src/Genesis/examples/tutorials/hello_genesis.py
free(): invalid pointer
Fatal Python error: Aborted

Current thread 0x00007f49ca598d80 (most recent call first):
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1173 in create_module
  File "<frozen importlib._bootstrap>", line 565 in module_from_spec
  File "<frozen importlib._bootstrap>", line 666 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1058 in _handle_fromlist
  File "/home/zhan/anaconda3/envs/genesis/lib/python3.9/site-packages/taichi/_lib/utils.py", line 46 in import_ti_python_core
  File "/home/zhan/anaconda3/envs/genesis/lib/python3.9/site-packages/taichi/_lib/utils.py", line 110 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 850 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/home/zhan/anaconda3/envs/genesis/lib/python3.9/site-packages/taichi/_lib/__init__.py", line 1 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 850 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/home/zhan/anaconda3/envs/genesis/lib/python3.9/site-packages/taichi/lang/impl.py", line 6 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 850 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1058 in _handle_fromlist
  File "/home/zhan/anaconda3/envs/genesis/lib/python3.9/site-packages/taichi/lang/__init__.py", line 1 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 850 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/home/zhan/anaconda3/envs/genesis/lib/python3.9/site-packages/taichi/_funcs.py", line 3 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 850 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/home/zhan/anaconda3/envs/genesis/lib/python3.9/site-packages/taichi/__init__.py", line 1 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 850 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/home/zhan/anaconda3/envs/genesis/lib/python3.9/site-packages/genesis/__init__.py", line 13 in <module>
  File "<frozen importlib._bootstrap>", line 228 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 850 in exec_module
  File "<frozen importlib._bootstrap>", line 680 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007 in _find_and_load
  File "/home/zhan/src/Genesis/examples/tutorials/hello_genesis.py", line 4 in <module>
Aborted (core dumped)`

this line " from taichi._lib.core import taichi_python as core  # pylint: disable=C0415
    except Exception as e:"

def import_ti_python_core():
    if get_os_name() != "win":
        # pylint: disable=E1101
        old_flags = sys.getdlopenflags()
        sys.setdlopenflags(2 | 8)  # RTLD_NOW | RTLD_DEEPBIND
    else:
        pyddir = os.path.dirname(os.path.realpath(__file__))
        os.environ["PATH"] += os.pathsep + pyddir
    try:
        from taichi._lib.core import taichi_python as core  # pylint: disable=C0415
    except Exception as e:
        if isinstance(e, ImportError):
            print(
                Fore.YELLOW + "Share object taichi_python import failed, "
                "check this page for possible solutions:\n"
[... cut ...] .org/docs/install" + Fore.RESET
            )
            if get_os_name() == "win":
                # pylint: disable=E1101
                e.msg += "\nConsider installing Microsoft Visual C++ Redistributable: https://aka.ms/vs/16/release/vc_redist.x64.exe"
        raise e from None

    if get_os_name() != "win":
        sys.setdlopenflags(old_flags)  # pylint: disable=E1101
    lib_dir = os.path.join(package_root, "_lib", "runtime")
    core.set_lib_dir(locale_encode(lib_dir))
    return core

closed 2025-05-24 by duburcqa