Genesis-Embodied-AI/genesis-world #298: how to get the coordinate for a soft robot?

zhang-qiang-github, 2024-12-25:
For a sphere:
```
robot_fem = scene.add_entity(
    morph=gs.morphs.Sphere(
        pos=(0.5, -0.2, 0.3),
        radius=0.1,
    ),
    material=gs.materials.FEM.Muscle(
        E=E,
        nu=nu,
        rho=rho,
        model='stable_neohooken',
    ),
)
```

It consist of many points and the sphere would elastic deformation. How can I get the coordinate for all points in the simulation?

zhouxian, 2024-12-25:
@zswang666 can you take a look

zhanglepy, 2024-12-26:
hello ， I am also very concerned about this issue

zswang666, 2024-12-27:
```python
state = robot_fem.get_state()
pos = state.pos # positions of all vertices in the elements
vel = state.vel # velocities of all vertices in the elements
```

closed 2024-12-27 by zswang666