Genesis-Embodied-AI/genesis-world #304: The problem of using the function of gs.RigidJoint 'get_pos()'

thewidesky, 2024-12-25:
left_feet_uid = H1.get_joint(name=feet_name[0]).uid
print("uid:",left_feet_uid)

type = H1.get_joint(name=feet_name[0]).type
print('type is:',type)

left_feet_pos = H1.get_joint(name=feet_name[0]).get_pos()
print("pos is:",left_feet_pos)`

having the problem:
Traceback (most recent call last):
  File "/home/xiaojunjie/genesis_h1/Simple_h1.py", line 74, in <module>
    left_feet_pos = H1.get_joint(name=feet_name[0]).get_pos()
  File "/home/xiaojunjie/miniconda3/envs/genesis/lib/python3.9/site-packages/genesis/utils/misc.py", line 48, in wrapper
    return method(self, *args, **kwargs)
  File "/home/xiaojunjie/miniconda3/envs/genesis/lib/python3.9/site-packages/genesis/engine/entities/rigid_entity/rigid_joint.py", line 84, in get_pos
    self._kernel_get_pos(tensor)
  File "/home/xiaojunjie/miniconda3/envs/genesis/lib/python3.9/site-packages/taichi/lang/kernel_impl.py", line 1178, in __call__
    raise type(e)("\n" + str(e)) from None
taichi.lang.exception.TaichiIndexError: 
File "/home/xiaojunjie/miniconda3/envs/genesis/lib/python3.9/site-packages/genesis/engine/entities/rigid_entity/rigid_joint.py", line 93, in _kernel_get_pos:
            l_info = self._solver.links_info[self._idx, i_b]
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Field with dim 1 accessed with indices of dim 2

How can I solve this problem?
By the way, did somebody know how to make H1 walking using the genesis? I am glad to here the answer.

zhouxian, 2024-12-25:
yes we have made h1 walking in genesis

@ziyanx02 maybe we can release that ckpt?

zhouxian, 2024-12-25:
btw I have assigned @Kashu7100 to fix this bug

Kashu7100, 2024-12-25:
@thewidesky It would be helpful for me to have the code snippet and assets that you used for debugging!

thewidesky, 2024-12-26:
If you can make H1 walking as a example to the official file,it would be great for the learner of the genesis.

The code I write is: (URDF from the Unitree official github)

import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
H1 = scene.add_entity(
    gs.morphs.URDF(file = 'h1.urdf',
                   pos = (0,0,1))
)
scene.build()
jnt_names = [
    'left_hip_yaw_joint',
    'left_hip_roll_joint',
    'left_hip_pitch_joint',
    'left_knee_joint',
    'left_ankle_joint',
    'right_hip_yaw_joint',
    'right_hip_roll_joint',
    'right_hip_pitch_joint',
    'right_knee_joint',
    'right_ankle_joint',
    'torso_joint',
    'left_shoulder_pitch_joint',
    'left_shoulder_roll_joint',
    'left_shoulder_yaw_joint',
    'left_elbow_joint',
    'right_shoulder_pitch_joint',
    'right_shoulder_roll_joint',
    'right_shoulder_yaw_joint',
    'right_elbow_joint'
]
feet_name = ['left_ankle_joint','right_ankle_joint']
feet_idx = [H1.get_joint(name).dof_idx_local for name in feet_name]

left_feet_uid = H1.get_joint(name=feet_name[0]).uid
print("uid:",left_feet_uid)
type = H1.get_joint(name=feet_name[0]).type
print('type is:',type)
left_feet_pos = H1.get_joint(name=feet_name[0]).get_pos()
print("pos is:",left_feet_pos)

for i in range(5):
    scene.step()

thewidesky, 2024-12-26:
@Kashu7100 @zhouxian

Kashu7100, 2024-12-26:
Thanks! Are you using this URDF (https://github.com/unitreerobotics/unitree_rl_gym/tree/main/resources/robots/h1)?
Also, could you enclose your code in a code block (```code```) next time?

thewidesky, 2024-12-26:
URDF(https://github.com/unitreerobotics/unitree_ros/tree/master/robots/h1_description/urdf/h1.urdf).
I will use the code block next time.`

Kashu7100, 2024-12-26:
Thanks. I think there was a bug in `rigid_joint.py`. I created a PR https://github.com/Genesis-Embodied-AI/Genesis/pull/323

zhouxian, 2024-12-27:
@Kashu7100 @zswang666 i think this is still unfixed?

zswang666, 2024-12-27:
Hmm...it seems like the issue should be fixed as in [rigid_joint._kernel_get_pos](https://github.com/Genesis-Embodied-AI/Genesis/blob/c7200f59213aeab4abf45288c87b7dce1d96feec/genesis/engine/entities/rigid_entity/rigid_joint.py#L93) and [rigid_joint._kernel_get_quat](https://github.com/Genesis-Embodied-AI/Genesis/blob/c7200f59213aeab4abf45288c87b7dce1d96feec/genesis/engine/entities/rigid_entity/rigid_joint.py#L128).

closed 2024-12-27 by zswang666

Kashu7100, 2024-12-28:
I think this should be fine.