Genesis-Embodied-AI/genesis-world #307: Is there a way to know how many dof from a xml file?

milong26, 2024-12-25:
As title mentioned,I want to know how many joints and dof does franka_sim/bi-franka_panda have,to let it move.

 But it didn't say in its github page or readme.md. So is there any solution?

Thanks!

milong26, 2024-12-25:
By the way if anyone know standard dataset of robots(especially two arm) plz tell me :sob:

zhouxian, 2024-12-25:
Do you mean inside genesis or by just reading the xml?

In genesis, you can simply do `entity.n_dofs`, or simply type `entity` in a interactive shell (like ipython)
see https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/interactive_debugging.html

Kashu7100, 2024-12-25:
Also, you can use `gs view <path to file>` to check the asset and moveable dof

milong26, 2024-12-26:
ty for telling me info about inside genesis,how about reading the xml? SInce I may need to import objects from other dataset.

YilingQiao, 2025-02-23:
You can read the XML using Genesis and access its DoFs as mentioned in the answer above. If you do not want to load it into Genesis, you can also use MuJoCo's API to load the XML and retrieve the DoFs from MuJoCo.

closed 2025-02-23 by YilingQiao