Genesis-Embodied-AI/genesis-world #309: Environment ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies.

yjwang346, 2024-12-25:
I am using a MacBook with an Apple M2 Pro chip. First, I installed PyTorch using the command conda install pytorch::pytorch torchvision torchaudio -c pytorch. Then, when I tried to execute the command pip install genesis-world, I encountered an error.

Collecting numba (from genesis-world)
  Using cached numba-0.60.0-cp39-cp39-macosx_11_0_arm64.whl.metadata (2.7 kB)
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
Collecting genesis-world
  Using cached genesis_world-0.1.1-py3-none-any.whl.metadata (11 kB)
ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.0 depends on pymeshlab
    genesis-world 0.1.1 depends on pymeshlab

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
[image]

yjwang346, 2024-12-26:
And when I recreate a new environment, and firstly run commond "pip install genesis-world", I encountered the error: "ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies."

(The version of python is 3.9)

gemcollector, 2024-12-26:
I encountered the same problem

Yao-Xinchen, 2024-12-28:
I solved this problem by recreating an environment with python 3.10.

```
conda create --name genesis python==3.10
conda activate genesis
pip install genesis-world
```

I am using MacBook Air M2.

ChristpherHan, 2024-12-28:
It's because PyMeshLab has not been tested in a Conda environment. Please visit https://github.com/cnr-isti-vclab/PyMeshLab. I solved this by downloading the source package and installing it.

thy950523, 2025-01-03:
> It's because PyMeshLab has not been tested in a Conda environment. Please visit https://github.com/cnr-isti-vclab/PyMeshLab. I solved this by downloading the source package and installing it.

it really works for me, thanks.

closed 2025-02-23 by YilingQiao