Genesis-Embodied-AI/genesis-world #310: Windows 11 - No Camera Support?

skurtyyskirts, 2024-12-25:
I've been trying to launch an example script with genesis but get a "Camerea not yet supported on Windows". I believe I read somewhere genesis was supported for Windows, but rendering isn't yet? Just making sure that this is correct for the error I'm receiving. Also - would it be possible to use genesis in a virtual machine? 
```
Microsoft Windows [Version 10.0.22631.4602]
(c) Microsoft Corporation. All rights reserved.

C:\Users\skurtyy\Desktop>python lol.py
[Genesis] [08:06:03] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [08:06:03] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [08:06:03] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [08:06:03] [INFO] Running on [Intel64 Family 6 Model 183 Stepping 1, GenuineIntel] with backend gs.cpu. Device memory: 63.72 GB.
[Genesis] [08:06:03] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [08:06:04] [INFO] Scene <03f1117> created.
[Genesis] [08:06:04] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <5448df9>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [08:06:04] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <a992562>, morph: <gs.morphs.MJCF(file='C:\Users\skurtyy\AppData\Local\Programs\Python\Python311\Lib\site-packages\genesis\assets\xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
Traceback (most recent call last):
  File "C:\Users\skurtyy\Desktop\lol.py", line 38, in <module>
    cam = scene.add_camera(
          ^^^^^^^^^^^^^^^^^
  File "C:\Users\skurtyy\AppData\Local\Programs\Python\Python311\Lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\skurtyy\AppData\Local\Programs\Python\Python311\Lib\site-packages\genesis\engine\scene.py", line 444, in add_camera
    return self._visualizer.add_camera(res, pos, lookat, up, model, fov, aperture, focus_dist, GUI, spp, denoise)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\skurtyy\AppData\Local\Programs\Python\Python311\Lib\site-packages\genesis\vis\visualizer.py", line 72, in add_camera
    gs.raise_exception("Camera not yet supported on Windows but is supposed to. Can anyone help?")

[Genesis] [08:06:04] [ERROR] Camera not yet supported on Windows but is supposed to. Can anyone help?
[Genesis] [08:06:04] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

zhouxian, 2024-12-25:
Sadly we haven't figured out rendering on windows, as indicated in the installation page in the documentation...
We just released a linux docker, hope that would help

skurtyyskirts, 2024-12-26:
Oh that's excellent, thank you! I'll check that out, best wishes on the projects, this is absolutely incredible.

closed 2024-12-26 by skurtyyskirts