Genesis-Embodied-AI/genesis-world #312: RuntimeError: Validation failure (0): active samplers with a different type refer to the same texture image unit

RoninsCode, 2024-12-25:
I attempted to run the hello_genesis.py simulation provided with Genesis 0.2.0. During the scene creation and activation of the viewer, an OpenGL error occurs:

Error message:

RuntimeError: Validation failure (0): b'active samplers with a different type refer to the same texture image unit'

Expected Behavior:

The simulation should execute without rendering errors, and the viewer should display the scene correctly.
Reproduction

Code Example: hello_genesis.py

        python3 hello_genesis.py

Error Output:

            RuntimeError: Validation failure (0): b'active samplers with a different type refer to the same texture image unit'

System Information

    Operating System: Ubuntu 22.04 (64-bit)
    Python Version: 3.10.12
    Genesis Version: 0.2.0
    GPU: AMD Radeon RX 570
    GPU Driver: AMDGPU open-source driver (Kernel module: amdgpu)
    OpenGL Version: 4.6 (Mesa 24.3.2 - kisak-mesa PPA)

Troubleshooting Steps Already Attempted

Driver Check:
       
        lspci -k | grep -EA3 'VGA|3D'

Output:

     07:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere [Radeon RX 
     470/480/570/570X/580/580X/590] (rev e7)
    Subsystem: ASUSTeK Computer Inc. Ellesmere [Radeon RX 470/480/570/570X/580/580X/590]
    Kernel driver in use: amdgpu
    Kernel modules: amdgpu

Tested Software Rendering:

    MESA_LOADER_DRIVER_OVERRIDE=llvmpipe LIBGL_ALWAYS_SOFTWARE=1 python3 hello_genesis.py

The error persists.

    Updated Dependencies:
        pyglet: Version 1.5.27
        PyOpenGL: Version 3.1.6
        PyOpenGL-accelerate: Version 3.1.6

Request for Help

Could you advise how to resolve this issue or disable rendering temporarily? Let me know if additional logs or details are needed.

Knowledge-Stars, 2024-12-28:
I meet the same problem in Ubuntu22.04 ,seems like a problem in openGL?

RoninsCode, 2024-12-28:
import os
     os.environ['PYOPENGL_PLATFORM'] = 'glx'

add this before the genesis import

closed 2024-12-28 by RoninsCode

Sijiahui, 2024-12-30:
> ```
>  import os
>  os.environ['PYOPENGL_PLATFORM'] = 'glx'
> ```
> 
> 在 Genesis 导入之前添加这个，对我有用

I have used this method, but it doesn't work.

[image]

RoninsCode, 2024-12-30:
Can u show me the command: glxinfo | grep "OpenGL version"

Knowledge-Stars, 2024-12-31:
I upgraded my OpenGL to 3.1.7 and the problem solved , don't upgrade Pyrender , its version : 0.1.45.

wjz050726, 2025-01-04:
> I upgraded my OpenGL to 3.1.7 and the problem solved , don't upgrade Pyrender , its version : 0.1.45.

pyrender 0.1.45 requires PyOpenGL==3.1.0.pyopengl 3.1.7 is incompatible for me.  still got the error RuntimeError: Validation failure (0): b'active samplers with a different type refer to the same texture image unit'       :(

randarura, 2025-01-08:
I ran into the same problem. I reviewed the installation from the beginning.
I did not need pyrender, it worked fine with PyOpenGL==3.1.7.

absl-py==2.1.0
annotated-types==0.7.0
black==24.10.0
certifi==2024.12.14
charset-normalizer==3.4.1
click==8.1.8
coacd==1.0.5
colorama==0.4.6
contourpy==1.3.1
cycler==0.12.1
dataclasses-json==0.6.7
decorator==5.1.1
Deprecated==1.2.15
dill==0.3.9
etils==1.11.0
filelock==3.13.1
fonttools==4.55.3
freetype-py==2.5.1
fsspec==2024.2.0
genesis-world==0.2.1
glfw==2.8.0
idna==3.10
imageio==2.36.1
imageio-ffmpeg==0.5.1
importlib_resources==6.5.2
Jinja2==3.1.3
kiwisolver==1.4.8
lazy_loader==0.4
libigl==2.5.1
llvmlite==0.43.0
lxml==5.3.0
markdown-it-py==3.0.0
MarkupSafe==2.1.5
marshmallow==3.24.1
matplotlib==3.10.0
mdurl==0.1.2
moviepy==2.1.1
mpmath==1.3.0
mujoco==3.2.5
mypy-extensions==1.0.0
networkx==3.2.1
numba==0.60.0
numpy==1.26.4
nvidia-cublas-cu12==12.4.5.8
nvidia-cuda-cupti-cu12==12.4.127
nvidia-cuda-nvrtc-cu12==12.4.127
nvidia-cuda-runtime-cu12==12.4.127
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.2.1.3
nvidia-curand-cu12==10.3.5.147
nvidia-cusolver-cu12==11.6.1.9
nvidia-cusparse-cu12==12.3.1.170
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.4.127
nvidia-nvtx-cu12==12.4.127
opencv-python==4.10.0.84
OpenEXR==3.3.2
packaging==24.2
pathspec==0.12.1
pillow==10.2.0
platformdirs==4.3.6
plotly==5.24.1
pooch==1.8.2
proglog==0.1.10
psutil==6.1.1
pycollada==0.8
pydantic==2.7.1
pydantic_core==2.18.2
PyGEL3D==0.5.2
pyglet==2.0.20
pygltflib==1.16.0
Pygments==2.19.1
pymeshlab==2023.12.post2
PyOpenGL==3.1.7
pyparsing==3.2.1
python-dateutil==2.9.0.post0
python-dotenv==1.0.1
pyvista==0.44.2
requests==2.32.3
rich==13.9.4
scikit-image==0.25.0
scipy==1.15.0
scooby==0.10.0
screeninfo==0.8.1
six==1.17.0
sympy==1.13.1
taichi==1.7.3
tenacity==9.0.0
tetgen==0.6.4
tifffile==2024.12.12
tomli==2.2.1
torch==2.5.1+cu124
torchaudio==2.5.1+cu124
torchvision==0.20.1+cu124
tqdm==4.67.1
triton==3.1.0
typing-inspect==0.9.0
typing_extensions==4.9.0
urllib3==2.3.0
vtk==9.3.1
wrapt==1.17.0
zipp==3.21.0