Genesis-Embodied-AI/genesis-world #313: problem with requirement taichi==1.7.2 on Mac Anaconda Python 3.13.1 env

ai-bits, 2024-12-25:
Mac M3 36GB latest OS Anaconda Python 3.13.1 env, both `pip install -e .` and `pip install genesis-world` stumble over taichi==1.7.2 

```
(genesis) max:Genesis gy$ pip install -e .
Obtaining file:///Users/gy/dl/Genesis
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting psutil (from genesis-world==0.2.0)
  Downloading psutil-6.1.1-cp36-abi3-macosx_11_0_arm64.whl.metadata (22 kB)
Collecting scikit-image (from genesis-world==0.2.0)
  Downloading scikit_image-0.25.0-cp313-cp313-macosx_12_0_arm64.whl.metadata (14 kB)
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement taichi==1.7.2 (from genesis-world) (from versions: none)
ERROR: No matching distribution found for taichi==1.7.2
(genesis) max:Genesis gy$ pip install genesis-world
Collecting genesis-world
  Downloading genesis_world-0.2.0-py3-none-any.whl.metadata (11 kB)
Collecting psutil (from genesis-world)
  Using cached psutil-6.1.1-cp36-abi3-macosx_11_0_arm64.whl.metadata (22 kB)
Collecting scikit-image (from genesis-world)
  Using cached scikit_image-0.25.0-cp313-cp313-macosx_12_0_arm64.whl.metadata (14 kB)
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
Collecting genesis-world
  Downloading genesis_world-0.1.1-py3-none-any.whl.metadata (11 kB)
ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.0 depends on taichi==1.7.2
    genesis-world 0.1.1 depends on taichi==1.7.2
```
Will try on Xeon 3435 256GB 2 x 20GB RTX 4000 Ada Ubuntu 24.04 next if things work there.

Merry Xmas and thanks for this what hopefully turns out a real pearl soon
G.

zanxueyan, 2024-12-26:
genesis-world does require python3.9 or higher, but the taichi dependent can only be python3.8 at most, so I don't know how to solve this problem.
ERROR: Could not find a version that satisfies the requirement taichi==1.7.2 (from versions: none)
ERROR: No matching distribution found for taichi==1.7.2

ai-bits, 2024-12-26:
Thanks for trying!
I actually wanted to go for install on Ubuntu x64, but fell into Windows 11 on the above mentioned Xeon / Nvidia machine.
Same problem for taichi==1.7.2.
Thought I go ahead with PyTorch
conda install pytorch torchvision torchaudio pytorch-cuda=12.4 -c pytorch -c nvidia

Throws a gazillion of conflicts.
Firstly torchaudio is only compatible with <3.13.0a0, so >=3.12
...
Went back to the M3 MBP and tried PyTorch.
First stumbling block: Torchvision on Mac wants >=3.10 <3.110a0 (=3.11alpha, means some 3.10)

Could you update these two basic insights in README!?
Gonna retry on Mac and Win x64 and try on Ubuntu 24.04 later today CET.
Thanks again
G.

duburcqa, 2025-03-22:
Python requirements have been updated accordingly.

closed 2025-03-22 by duburcqa