Genesis-Embodied-AI/genesis-world #317: IPython scene return TypeError: __repr__ returned non-string (type NoneType)

milong26, 2024-12-26:
I have searched from issues no found same problem.

I followed this [Interactive Information Access and Debugging](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/interactive_debugging.html#interactive-information-access-and-debugging)

ipython --version=8.31.0

debug.py:
```python
import genesis as gs

gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=False)

plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

cam_0 = scene.add_camera()
scene.build()

# enter IPython's interactive mode
import IPython; IPython.embed()
```

log is here:

```
python debug.py 
Python 3.10.0 (default, Mar  3 2022, 09:58:08) [GCC 7.5.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.31.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: scene
Out[1]: ---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
File ~/anaconda3/envs/genesis/lib/python3.10/site-packages/IPython/core/formatters.py:770, in PlainTextFormatter.__call__(self, obj)
    763 stream = StringIO()
    764 printer = pretty.RepresentationPrinter(stream, self.verbose,
    765     self.max_width, self.newline,
    766     max_seq_length=self.max_seq_length,
    767     singleton_pprinters=self.singleton_printers,
    768     type_pprinters=self.type_printers,
    769     deferred_pprinters=self.deferred_printers)
--> 770 printer.pretty(obj)
    771 printer.flush()
    772 return stream.getvalue()

File ~/anaconda3/envs/genesis/lib/python3.10/site-packages/IPython/lib/pretty.py:419, in RepresentationPrinter.pretty(self, obj)
    408                         return meth(obj, self, cycle)
    409                 if (
    410                     cls is not object
    411                     # check if cls defines __repr__
   (...)
    417                     and callable(_safe_getattr(cls, "__repr__", None))
    418                 ):
--> 419                     return _repr_pprint(obj, self, cycle)
    421     return _default_pprint(obj, self, cycle)
    422 finally:

File ~/anaconda3/envs/genesis/lib/python3.10/site-packages/IPython/lib/pretty.py:794, in _repr_pprint(obj, p, cycle)
    792 """A pprint that just redirects to the normal repr function."""
    793 # Find newlines and replace them with p.break_()
--> 794 output = repr(obj)
    795 lines = output.splitlines()
    796 with p.group():

TypeError: __repr__ returned non-string (type NoneType)

```

zhouxian, 2024-12-26:
@zhenjia-xu Could you verify if this works?

zhouxian, 2024-12-27:
see #319

closed 2024-12-27 by zhouxian