Genesis-Embodied-AI/genesis-world #318: Segmentation fault (core dumped)

coghks625, 2024-12-26:
I encounter the same issue regardless of the installation method, whether it's:

Installing via pip install genesis-world, or
Cloning the repository with:
bash
Copy code
git clone https://github.com/Genesis-Embodied-AI/Genesis.git
cd Genesis
pip install -e .
When running the script ik_motion_planning_grasp.py, I consistently get a Segmentation fault error. However, this issue does not occur when executing the script inside a Docker container, where it works perfectly.

Here are the details of my environment:

OS: Ubuntu 20.04
cuda: 12.4
I replicated the installation process described in the Dockerfile on my local machine, step by step, but the result remains the same.
I even exported the Python package list from the Docker container and installed the exact same packages in a local virtual environment, yet the segmentation fault persists.
Additionally, I always get the following warning, regardless of whether I run the script in Docker or locally:

css
Copy code
[Genesis] [13:21:20] [WARNING] `qpos_start` exceeds joint limit. Relaxing joint limit to contain `qpos_start` for planning.
Questions:
Why does the script work flawlessly inside the Docker container but fails with a segmentation fault on my local machine, even with an identical Python environment?
What does the warning message mean, and is it indicative of a potential issue?
Thank you for your help!

HaodongHuang-W, 2024-12-30:
I encountered the same problem when running reinforcement learning cases, where I encountered a segmentation fault (core dummy)

bxtbold, 2025-01-13:
If you have OMPL compiled in your system (e.g. ros-\<distro\>-ompl), that is prolly caused by conflicted libraries. You can refer to #505

HaodongHuang-W, 2025-01-13:
@bxtbold Hello, may I ask if I also encountered a Segmentation fault (core dumped) while training go2's localization? Is this the same issue?

bxtbold, 2025-01-13:
> @bxtbold Hello, may I ask if I also encountered a Segmentation fault (core dumped) while training go2's localization? Is this the same issue?

I am not really sure since I am not provided the info. You may wanna open a new issue with some more details.

HaodongHuang-W, 2025-01-13:
@bxtbold  thx!

coghks625, 2025-01-14:
> If you have OMPL compiled in your system (e.g. ros-<distro>-ompl), that is prolly caused by conflicted libraries. You can refer to #505

wow i remove source /opt/ros/foxy/setup.bash in ~/.bashrc  segmentation fault is solved
my problem is conflict with ros2

closed 2025-01-14 by coghks625