Genesis-Embodied-AI/genesis-world #320: Add camera and no image

kelo020304, 2024-12-26:
[image]
when I am doing RL, I add a camea and rander it but i can not get a image , code below:
```python
self.cam = self.scene.add_camera(model='pinhole',
                                         res=(640, 480),
                                         pos=(0,0,1),
                                         lookat=(5,0,1),
                                         fov = 40,
                                         GUI=True)
        self.scene.build(n_envs= self.num_envs)
        # self.cam_pos = self.Piper.get_links_pos()[:, self.Piper.get_link('link6').idx - 1].cpu().numpy()
        # for i in range(self.num_envs):
        #     single_cam_pos = self.Piper.get_links_pos()[i, self.Piper.get_link('link6').idx - 1].cpu().numpy()
        #     self.cam.set_pose(pos=single_cam_pos)
        cv2.waitKey(1)
        self.cam.render()
``` 
How to solve it?

Genesis-Embodied-AI, 2024-12-26:
waitkey(1) should be after `render()`.

Could you check the variables returned by `render()` and see if it's meaingful?

kelo020304, 2024-12-26:
> waitkey(1) should be after `render()`.
> 
> Could you check the variables returned by `render()` and see if it's meaingful?

I have tried but it also don not work , render() return a 4 length tuple and the first element seems meaningful with shape (320,320,3) but also nothing to see

zhenjia-xu, 2025-01-10:
The camera pos is (0,0,1), is this inside the robot? Could you please change camera pos and give it a try?

YilingQiao, 2025-02-23:
Closing the issue due to no response. We can revisit it if a new object can be provided.

closed 2025-02-23 by YilingQiao