Genesis-Embodied-AI/genesis-world #326: when building a docker image, it always fails at RUN chmod +x ./build_luisa.sh && ./build_luisa.sh ${PYTHON_VERSION}

zhrli, 2024-12-26:
I can not take a snapshot.   

Anyone meets the same situation?

Shikairan, 2024-12-26:
more infomation plz
多点细节内容啊老哥

Kashu7100, 2024-12-26:
Can you provide me your system information (hardware specs, cuda version, etc.) and the exact error message that you are getting?

zhrli, 2024-12-26:
> Can you provide me your system information (hardware specs, cuda version, etc.) and the exact error message that you are getting?
CPU:                     72
  在线 CPU 列表：        0-71
厂商 ID：                GenuineIntel
  型号名称：             Intel(R) Xeon(R) Platinum 8352V CPU @ 2.10GHz
    CPU 系列：           6
    型号：               106
    每个核的线程数：     2

GPU Info (if applicable):
Thu Dec 26 16:37:49 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.113.01             Driver Version: 535.113.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |

4090 两张

Operating System:
Linux ubuntu 6.2.0-26-generic #26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 13 16:27:29 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

CUDA Version:
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Jun_13_19:16:58_PDT_2023
Cuda compilation tools, release 12.2, V12.2.91
Build cuda_12.2.r12.2/compiler.32965470_0

Python and Libraries:
Python 3.11.7

When building luisa ， my computer just collapsed.   So, I don't have any info about it

zhrli, 2024-12-26:
> Can you provide me your system information (hardware specs, cuda version, etc.) and the exact error message that you are getting?

Can you provide me your system information (hardware specs, cuda version, etc.) and the exact error message that you are getting?
CPU: 72
在线 CPU 列表： 0-71
厂商 ID： GenuineIntel
型号名称： Intel(R) Xeon(R) Platinum 8352V CPU @ 2.10GHz
CPU 系列： 6
型号： 106
每个核的线程数： 2

GPU Info (if applicable):
Thu Dec 26 16:37:49 2024
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.113.01 Driver Version: 535.113.01 CUDA Version: 12.2 |
|-----------------------------------------+----------------------+----------------------+
| GPU Name Persistence-M | Bus-Id Disp.A | Volatile Uncorr. ECC |
| Fan Temp Perf Pwr:Usage/Cap | Memory-Usage | GPU-Util Compute M. |
| | | MIG M. |

4090 两张

Operating System:
Linux ubuntu 6.2.0-26-generic https://github.com/Genesis-Embodied-AI/Genesis/issues/26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 13 16:27:29 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

CUDA Version:
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Jun_13_19:16:58_PDT_2023
Cuda compilation tools, release 12.2, V12.2.91
Build cuda_12.2.r12.2/compiler.32965470_0

Python and Libraries:
Python 3.11.7

When building luisa ， my computer just collapsed. So, I don't have any info about it

Kashu7100, 2024-12-26:
Can you tell me which line of `build_luisa.sh` causing you the failure at least?
With the current information provided, I have no clue what is happening.

Shikairan, 2024-12-26:
> When building luisa ， my computer just collapsed. So, I don't have any info about it

你这段过了么，如果你在国内的话我怀疑是git没整完超时了
git submodule update --init --recursive

zhrli, 2024-12-26:
> > When building luisa ， my computer just collapsed. So, I don't have any info about it
> 
> 你这段过了么，如果你在国内的话我怀疑是git没整完超时了 git submodule update --init --recursive

过了，是超时了好几次。

zhrli, 2024-12-26:
> Can you tell me which line of `build_luisa.sh` causing you the failure at least? With the current information provided, I have no clue what is happening.

I just used a MAC to reproduce that.  CPU load is 100% full.

[image]

zhrli, 2024-12-26:
> Can you tell me which line of `build_luisa.sh` causing you the failure at least? With the current information provided, I have no clue what is happening.

and the command I use:     docker buildx build --platform linux/amd64 -t genesis -f docker/Dockerfile docker --push

Kashu7100, 2024-12-26:
What happens if you do not do the cross compile (without using buildx)?

zhrli, 2024-12-26:
> What happens if you do not do the cross compile (without using buildx)?

[image]

skurtyyskirts, 2024-12-26:
I also had this error message trying to build the docker on Windows 11. I created a pull request with the necessary changes, after making these I was successfully able to create Genesis! The issue was caused by the script having Windows-style line endings, which Linux couldn't understand. To fix this, I installed a tool to convert those line endings to a Linux-friendly format and made sure the script had the correct permissions to run. I also added a check to confirm the script was in the right place inside the Docker container before running it.

ningwenjie, 2024-12-26:
> 我在 Windows 11 上尝试构建 docker 时也遇到了此错误消息。我创建了一个包含必要更改的拉取请求，完成这些更改后，我成功创建了 Genesis！问题是由脚本具有 Windows 样式的行尾引起的，而 Linux 无法理解。为了解决这个问题，我安装了一个工具将这些行尾转换为 Linux 友好的格式，并确保脚本具有正确的运行权限。我还添加了一个检查，以确认脚本在运行之前位于 Docker 容器内的正确位置。

[image]
[image]

JosvanderWesthuizen, 2024-12-28:
Same here for me on Mac apple silicon.

```
(base) josvanderwesthuizen@mac Genesis % docker build -t genesis -f docker/Dockerfile docker
[+] Building 73.4s (8/26)                                                                                                                                       docker:desktop-linux
 => [internal] load build definition from Dockerfile                                                                                                                            0.0s
 => => transferring dockerfile: 4.51kB                                                                                                                                          0.0s
 => [internal] load metadata for docker.io/pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel                                                                                          0.8s
 => [internal] load .dockerignore                                                                                                                                               0.0s
 => => transferring context: 2B                                                                                                                                                 0.0s
 => [internal] load build context                                                                                                                                               0.0s
 => => transferring context: 1.70kB                                                                                                                                             0.0s
 => CACHED [builder  1/11] FROM docker.io/pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel@sha256:e8e63dd7baca894ba11fe1ba48a52a550793c8974f89b533d697784dd20a4dc0                   0.0s
 => => resolve docker.io/pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel@sha256:e8e63dd7baca894ba11fe1ba48a52a550793c8974f89b533d697784dd20a4dc0                                    0.0s
 => [builder  2/11] RUN apt-get update && apt-get install -y --no-install-recommends     build-essential     manpages-dev     libvulkan-dev     zlib1g-dev     xorg-dev libgl  60.3s
 => [stage-1  2/12] RUN apt-get update && apt-get install -y --no-install-recommends     tmux     git     curl     wget     bash-completion     libgl1     libgl1-mesa-glx     58.2s
 => [stage-1  3/12] WORKDIR /workspace                                                                                                                                          0.0s
 => [stage-1  4/12] RUN pip install --no-cache-dir open3d                                                                 [+] Building 73.6s (8/26)                                                                                                                                       docker:desktop-linux /opt/conda/lib/python3.11/site-packages (from pexpect>4.3->ipyt => [internal] load build definition from Dockerfile                                                                                                                            0.0snda/lib/python3.11/site-packages (from prompt-toolkit<3.1.0,>=3. => => transferring dockerfile: 4.51kB                                                                                                                                          0.0sn /opt/conda/lib/python3.11/site-packages (from stack-data->ipyt => [internal] load metadata for docker.io/pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel                                                                                          0.8sn /opt/conda/lib/python3.11/site-packages (from stack-data->ipyt => [internal] load .dockerignore                                                                                                                                               0.0sconda/lib/python3.11/site-packages (from stack-data->ipython>=6. => => transferring context: 2B                                                                                                                                                 0.0s_27_x86_64.whl (399.7 MB)                                        => [internal] load build context                                                                                                                                               0.0schain-r/test &&     apt update &&     apt install -y --no-instal => => transferring context: 1.70kB                                                                                                                                             0.0surity InRelease                                                  => CACHED [builder  1/11] FROM docker.io/pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel@sha256:e8e63dd7baca894ba11fe1ba48a52a550793c8974f89b533d697784dd20a4dc0                   0.0ste/cuda/repos/ubuntu2204/x86_64  InRelease                       => => resolve docker.io/pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel@sha256:e8e63dd7baca894ba11fe1ba48a52a550793c8974f89b533d697784dd20a4dc0                                    0.0slease                                                            => [builder  2/11] RUN apt-get update && apt-get install -y --no-install-recommends     build-essential     manpages-dev     libvulkan-dev     zlib1g-dev     xorg-dev libgl  60.3stes InRelease                                                    => [stage-1  2/12] RUN apt-get update && apt-get install -y --no-install-recommends     tmux     git     curl     wget     bash-completion     libgl1     libgl1-mesa-glx     58.2slchain-r/test/ubuntu jam [... cut ...]
  54 |     COPY build_luisa.sh /workspace/
  55 | >>> RUN ls -la /workspace && \
  56 | >>>     dos2unix /workspace/build_luisa.sh && \
  57 | >>>     chmod +x /workspace/build_luisa.sh && \
  58 | >>>     ./build_luisa.sh ${PYTHON_VERSION}
  59 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c ls -la /workspace &&     dos2unix /workspace/build_luisa.sh &&     chmod +x /workspace/build_luisa.sh &&     ./build_luisa.sh ${PYTHON_VERSION}" did not complete successfully: exit code: 2
```

JosvanderWesthuizen, 2024-12-28:
Same issue for me on Mac Apple silicon.
Error logs attached
[logs.txt](https://github.com/user-attachments/files/18264594/logs.txt)

Kashu7100, 2024-12-28:
Thank you for the error log. It seems the error comes from the incompatibility of the base pytorch image. I think we need to find an appropriate base image for arm based arch...

```
InvalidBaseImagePlatform: Base image pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel was pulled with platform "linux/amd64", expected "linux/arm64" for current build
```

zhrli, 2024-12-30:
> Can you tell me which line of `build_luisa.sh` causing you the failure at least? With the current information provided, I have no clue what is happening.

I think this problem is related to #380 . According to that solution, it works on linux.

AgentOfLyra, 2024-12-30:
@skurtyyskirts I used your dockerfile and still can't build, I've been stuck at this stage for a long time. 
[image]

rtmnkt, 2024-12-30:
Hello.
I was also getting a build error on the `ERROR [builder 11/11] RUN chmod +x ./build_luisa.sh && ./build_luisa.sh 3.11: 0.356 /bin/sh: 1: ./build_luisa.sh: not found` line. A PR may have already been created and the problem may be solved, but I ran the following command based on the conversation in this thread, and it seems to have succeeded in the build in my environment.
1. `git config --global core.autocrlf input` (I think `git config core.autocrlf false` will probably work too)
2. Delete the local `Genesis` repository directory and run git clone again. (Because the core.autocrlf option requires a git checkout)
  I tried this on the main branch (60704fd).

This is something that all Windows users need to do, so in reality it is probably better to control it with the `eol` in the `.gitattributes` file. (Unconfirmed)
This may be unnecessary information, but I hope it will be helpful. Sorry if it is not related to this case.

zhrli, 2024-12-30:
> @skurtyyskirts I used your dockerfile and still can't build, I've been stuck at this stage for a long time. [...]

#380

closed 2024-12-31 by zhrli