Genesis-Embodied-AI/genesis-world #328: Question about accessing IMU accelerometer data in Genesis

pjw971022, 2024-12-26:
I'm trying to simulate an IMU sensor in Genesis and have some questions about accelerometer data:

1. Is it possible to obtain accelerometer readings similar to a real IMU sensor in Genesis?

2. Currently, I can see that RigidSolver provides:
   - base_lin_vel (linear velocity)
   - base_ang_vel (angular velocity)
   - projected_gravity

   However, I can't find a direct way to calculate proper accelerometer readings, which should include:
   - Linear acceleration (rate of change of linear velocity)
   - Gravitational acceleration
   - Centrifugal effects

3. If accelerometer data is available, how can we access it? If not, what would be the recommended approach to implement IMU accelerometer simulation in Genesis?

zhouxian, 2024-12-29:
Yes it's definitely possible. Could you suggest some API format? We should be able to add it soon

pjw971022, 2025-01-02:
Thank you for your fast reply,

We just need the 3x1 vector information of the linear acceleration of any links (especially for the base of the robot)
Also is it possible to add the actual sensor modelling? (like adding noise, and the gravity vector values which is usually mixed in normal accelerometers?

Hope your fast reply :)

Best Regards,

Jinwoo Park.

zhouxian, 2025-01-02:
Yes I think we can add something similar to API for getting links velocity etc

@zswang666 could you help with this?

closed 2025-01-22 by zswang666