Genesis-Embodied-AI/genesis-world #330: how to deal with `qpos_start` exceeds joint limit. Relaxing joint limit to contain `qpos_start` for planning.

milong26, 2024-12-26:
I got these two warning:

[WARNING] `qpos_start` exceeds joint limit. Relaxing joint limit to contain `qpos_start` for planning.
[WARNING] `qpos_goal` exceeds joint limit. Relaxing joint limit to contain `qpos_goal` for planning.

And it looks like the path of robot hand is strange. :sad:

If you need my code: Im working on let franka_sim/bi-franka_panda to grasp a cude.
```python
import numpy as np
import genesis as gs

gs.init(backend=gs.cpu,logging_level='warning')

scene = gs.Scene(
    viewer_options = gs.options.ViewerOptions(
        camera_pos    = (3, -1, 1.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 30,
        max_FPS       = 60,
    ),
    show_viewer = True,
)
scene.add_entity(gs.morphs.Plane())

table = scene.add_entity(
     gs.morphs.Box(
         size = (0.4, 0.4, 0.4),
         pos  = (0, 0.4, 0.2),
         fixed=True

     )
 )
cube=scene.add_entity(
    gs.morphs.Box(
        size=(0.04,0.04,0.04),
        pos=(0,0.6,0.02),
    )
)

franka = scene.add_entity(
    gs.morphs.MJCF(
        file='xml/franka_sim/bi-franka_panda.xml',
        scale=0.5
    ),
)
scene.build()

motors_dof = np.arange(16)     
fingers_dof = np.arange(16, 18) 

end_effector1 = franka.get_link('panda0_gripper')
# end_effector2= franka.get_link('')

qpos = franka.inverse_kinematics(
    link = end_effector1,
    pos  = np.array([0, 0.6,0.7]),
    quat = np.array([0, 0, 1, 0]),
    respect_joint_limit=False
)
path = franka.plan_path(
    qpos_goal     = qpos,
    num_waypoints = 200, 
)

for waypoint in path:
    franka.control_dofs_position(waypoint)
    scene.step()
for i in range(100):
    scene.step()
for i in range(2000):
    scene.step()
```

stuPETER12138, 2024-12-26:
I got the same first warning when i run /tutorials/IK_motion_planning_grasp.py.

milong26, 2024-12-27:
> I got the same first warning when i run /tutorials/IK_motion_planning_grasp.py.

But tutorials/IK_motion_planning_grasp can complete the mission xD

milong26, 2024-12-27:
I'm curious about the `pos` in ik:
```python
qpos = franka.inverse_kinematics(
    link = end_effector,
    pos  = np.array([0.65, 0.0, 0.25]),
    quat = np.array([0, 1, 0, 0]),
)
path = franka.plan_path(
    qpos_goal     = qpos,
    num_waypoints = 200, # 2秒时长
)

for waypoint in path:
    franka.control_dofs_position(waypoint)
    scene.step()

for i in range(100):
    scene.step()
```

After this how can I get the position of end-effector which should be near [0.65, 0.0, 0.25]?

I tried end_effector.get_pos() but its output is far from [0.65, 0.0, 0.25] . :cry:

duburcqa, 2025-03-23:
This issue basically means that the position is out of reach, either because the solver fails to find a valid robot configuration or because it is really physically impossible. There is nothing Genesis itself can do about this. Closing.

closed 2025-03-23 by duburcqa