Genesis-Embodied-AI/genesis-world #335: Will more sensors be added soon?

JoeJoeOvO, 2024-12-26:
More sensors like lidar and cameras are urgently needed for simulations! It looks like only IMU has been added. Will more sensors be added soon?  And when will there be a more detailed introduction to the application of RL?

zhouxian, 2024-12-26:
camera is already there

Could you provide a reference example / API for lidar? That would be really helpful for us to make is as intuitive as other apis

oscar-lima, 2024-12-26:
This code implements a plugin for lidar simulation in Gazebo with GPU within ROS1 : https://github.com/ros-simulation/gazebo_ros_pkgs/blob/noetic-devel/gazebo_plugins/src/gazebo_ros_gpu_laser.cpp , many other sensors can be found here: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/noetic-devel/gazebo_plugins/src

closed 2025-05-24 by duburcqa