Genesis-Embodied-AI/genesis-world #336: [ERROR] No display detected. Use `show_viewer=False` for headless mode.

0ndyy, 2024-12-26:
I get this error every time, I only tried using the Hello World script, but no luck. I am using wsl2 on windows 10. In the .wslconfig guiapplications=true.

alanwang8, 2024-12-28:
Same issue here, I've only tried running the hello_genesis.py script also after installation

Kashu7100, 2024-12-29:
seems related to https://github.com/Genesis-Embodied-AI/Genesis/issues/227

Apande280, 2025-01-29:
Any progress? I keep having the issue and am not sure how to let Genesis detect a display.

Alberthor47, 2025-02-27:
I found this [post](https://www.datacamp.com/tutorial/genesis-physics-engine-tutorial) very useful, I was trying to run this hello-world example in Collab, and it seems that the problem is that Collab runs in a remote environment with no access to a local display.

closed 2025-02-28 by duburcqa

y9miao, 2025-08-29:
In my case, if I run the script within the terminal of VSCODE, it doesn't work. But it works with a normal terminal

duburcqa, 2025-08-30:
Could you post the entire traceback with `logging_level=“debug”` ?