Genesis-Embodied-AI/genesis-world #337: update genesis documentation about minimum Python version supported.

fbaptista, 2024-12-26:
pymeshlab is not able to install on Python 3.9, so this Python version is not supported. 

_Originally posted by @fbaptista in https://github.com/Genesis-Embodied-AI/Genesis/issues/52#issuecomment-2562906171_

zhouxian, 2024-12-26:
On linux it works fine with python 3.9 (this is the version we have been using for development)

which os are u referrring to?

fbaptista, 2024-12-26:
macOS / ARM.
See Issue #52 resolution for more details

ai-bits, 2024-12-26:
Genesis (from repo)
and PyTorch (`conda install pytorch torchvision torchaudio pytorch-cuda=12.4 -c pytorch -c nvidia` on Ubuntu 24.04 and Windows 11 latest)
install on a Python 3.10.16 env on an M3 MBP and a Xeon 256GB / Nvidia 2x RTX 4000.

But even the Hello `https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html` example fails on line 1 on all 3 platforms!

python py/genesis-hello.py
Traceback (most recent call last):
  File "/home/gy/dl/LinuxSetup/py/genesis-hello.py", line 1, in <module>
    import genesis as gs
  File "/home/gy/dl/Genesis/genesis/__init__.py", line 274, in <module>
    from .engine.scene import Scene
  File "/home/gy/dl/Genesis/genesis/engine/scene.py", line 5, in <module>
    from genesis.engine.entities import Emitter
  File "/home/gy/dl/Genesis/genesis/engine/entities/__init__.py", line 1, in <module>
    from .avatar_entity import AvatarEntity
  File "/home/gy/dl/Genesis/genesis/engine/entities/avatar_entity/__init__.py", line 1, in <module>
    from .avatar_entity import AvatarEntity
  File "/home/gy/dl/Genesis/genesis/engine/entities/avatar_entity/avatar_entity.py", line 3, in <module>
    from ..rigid_entity import RigidEntity
  File "/home/gy/dl/Genesis/genesis/engine/entities/rigid_entity/__init__.py", line 1, in <module>
    from .rigid_entity import RigidEntity
  File "/home/gy/dl/Genesis/genesis/engine/entities/rigid_entity/rigid_entity.py", line 8, in <module>
    from genesis.utils import mesh as mu
  File "/home/gy/dl/Genesis/genesis/utils/mesh.py", line 12, in <module>
    import tetgen
  File "/home/gy/anaconda3/envs/genesis/lib/python3.10/site-packages/tetgen/__init__.py", line 2, in <module>
    from tetgen import _tetgen
  File "tetgen/cython/tetgen/_tetgen.pyx", line 1, in init tetgen._tetgen
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

I think on the M3 torchaudio would run into problems with >3.10.
Thx
G.

fbaptista, 2024-12-26:
I can confirm that it works as expected with macOSARM + Python 3.12

I can make some tests on 3.10 if needed

ai-bits, 2024-12-26:
Thanks for helping!
Wondering  what was wrong on all 3 platforms on 3.10.16
(`File "tetgen/cython/tetgen/_tetgen.pyx", line 1, in init tetgen._tetgen
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject`)
I began with the recommended platform (Ubuntu), removed the env and reinstalled based on 3.12.8 and latest repo and PyTorch.
Same error.

You are right re 3.12 and Genesis and PyTorch.
Both installed contrary to what I had anticipated from the failed 3.13.1 trial messages.
The 3.13.1 install seems to have confused versions saying Python >=3.10, <3.11.0a0 was needed for torchvision. Guess the alpha version screwed up the comparison.

Any idea how to deal with the ValueError?
Should I move to a different thread, because this has nothing to do any more with your 3.9 problem?
Thx
G.

fbaptista, 2024-12-27:
Yes, this seems to be a different issue, the problem on Python 3.9 + macOS ARM is that we can't even install `pymeshlab` dependency.

You rissue can be due to `mismatch between the NumPy version used to compile the Cython extension and the version currently installed in your Python environment` ...

maybe updating or reinstalling the package? 
pip install --upgrade numpy

ai-bits, 2024-12-27:
Thanks again for chiming in!
Your suspicion / intuition is not far from what I found. See [docs Python -V must be >=3.9 AND <3.13 and install PyTorch before Genesis fixes numpy ver problems](https://github.com/Genesis-Embodied-AI/Genesis/issues/354)

Felix-Zhenghao, 2024-12-29:
Same issue. On macOS 14, pymeshlab can't be installed when using python 3.9. Problem solved when using python 3.10.

ai-bits, 2024-12-29:
@zhouxian just agreed on my above thread to change the docs to Python >=3.10 and <3.13 to account for your pymeshlab problem and torchvision had with 3.13.1 in my first attempt.
Hope he will also change suggested install order to PyTorch, THEN Genesis so torch won't overwrite Genesis' 1.26 numpy with the incompatible v2.

Cheers
G.

duburcqa, 2025-03-23:
Documentation has been updated. It will be reflected at next release.

closed 2025-03-23 by duburcqa