Genesis-Embodied-AI/genesis-world #340: Adding a light to the scene from a Sphere does not work

kaspar-gestalt, 2024-12-26:
```
Traceback (most recent call last):                                                                                                                                                     
  File "/home/user/python_scripts/genesis_rendering.py", line 42, in <module>                                                                                                        
    light_1 = scene.add_light(gs.morphs.Sphere(pos=(0,0,5), radius=0.1), color=(1,1,1,1), intensity=20)                                                                                
  File "/home/user/repos/Genesis/genesis/utils/misc.py", line 38, in wrapper                                                                                                         
    return method(self, *args, **kwargs)                                                                                                                                               
  File "/home/user/repos/Genesis/genesis/engine/scene.py", line 400, in add_light                                                                                                    
    self.visualizer.raytracer.add_mesh_light(                                                                                                                                          
  File "/home/user/repos/Genesis/genesis/vis/raytracer.py", line 200, in add_mesh_light                                                                                              
    name=str(mesh._id),    
```
Patched by adding an _id attribute in function definition of add_light: `mesh._id = 'testID'`

Progolue, 2025-01-08:
#480 
the issue is fixed

closed 2025-01-08 by Kashu7100

kaspar-gestalt, 2025-01-08:
@Progolue where?

Progolue, 2025-01-08:
> @Progolue where?

https://github.com/Genesis-Embodied-AI/Genesis/pull/494
To be more specific, change name=str(mesh._id) to name=str(mesh.uid).