Genesis-Embodied-AI/genesis-world #341: "Generating 4D dynamical & physical world" is this feature already in the code or coming soon?

sergeybok, 2024-12-26:
> Genesis's physics engine is empowered by a VLM-based generated agent that uses the APIs provided by the simulation infrastructure as tools to create 4D dynamic worlds, which can then be used as a foundational data source for extracting various modalities of data. Together with modules for generating camera and object motion, we are able to generate physically-accurate and view-consistent videos and other modalities of data.

I am a bit confused by this. There is nothing in the documentation about this but it's on the advertisement page. For example, the video of the World of Warcraft looking character running on the table -- is that created by assembling the assets and loading them as textured mesh (which I also didn't see support for that but maybe I misunderstood) along with the animation? Is the texture added by the Sim2Real step that you run after rendering the initial simulation? Or am I completely on the wrong track here? What is the VLM that they are talking about?

jmantherobot, 2024-12-27:
Yea, i've also been searching the repo for where anything generative happens and so far I only see normal engine stuff.  The only mention of, "Generative" i could find was in the roadmap 
[image]
where they put it basically as a TODO

jmantherobot, 2024-12-27:
I also saw https://github.com/Genesis-Embodied-AI/Genesis/issues/286 which asks basically the same question.

zhouxian, 2024-12-27:
see #6

closed 2024-12-27 by zhouxian