Genesis-Embodied-AI/genesis-world #343: Failing to run Genesis on Windows 11 & WSL

vstavros02, 2024-12-26:
Every time I try to run a scene from the 'Hello, Genesis' tutorial, it fails to load. I ensured that I have the prerequisites installed (such as PyTorch ans Conda) however I receive the same error . I've tried on my regular Windows 11 and WSL, having the same error. Does anyone know how to solve this? Is there something I'm missing, a dependency that needs to be installed? I've enured my GPU driver is correct and that everything is up to date and compatible! Any advice would help, thanks so much!

WSL
"(genesis_project) vickistavros@Timmy:~/genesis_code$ python
Python 3.11.11 (main, Dec 11 2024, 16:28:39) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.

import genesis as gs
kend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
scene.step()

gs.init(backend=gs.cpu)
[Genesis] [17:31:47] [INFO] ╭───────────────────────────────────────╮
[Genesis] [17:31:47] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [17:31:47] [INFO] ╰───────────────────────────────────────╯
[Genesis] [17:31:47] [INFO] Running on [12th Gen Intel(R) Core(TM) i9-12900HK] with backend gs.cpu. Device memory: 7.76 GB.
[Genesis] [17:31:47] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.

scene = gs.Scene(show_viewer=True)
[Genesis] [17:31:48] [INFO] Scene <7d95c05> created.
plane = scene.add_entity(gs.morphs.Plane())
[Genesis] [17:31:48] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <1e89bb6>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
franka = scene.add_entity(
... gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
... )
[Genesis] [17:31:48] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <6768ea0>, morph: <gs.morphs.MJCF(file='/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.

scene.build()
[Genesis] [17:31:48] [INFO] Building scene <7d95c05>...
[Genesis] [17:31:57] [INFO] Compiling simulation kernels...
[Genesis] [17:32:28] [INFO] Building visualizer...
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
self.run()
File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/threading.py", line 982, in run
self._target(*self._args, **self._kwargs)
File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 1126, in _init_and_start_app
super(Viewer, self).init(
File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/pyglet/window/xlib/init.py", line 168, in init
super().init(*args, **kwargs)
File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/pyglet/window/init.py", line 527, in init
context = config.create_context(gl.current_context)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/pyglet/gl/xlib.py", line 114, in create_context
return XlibContext(self, share)
^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/pyglet/gl/xlib.py", line 149, in init
raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context"

Windows 11
:"Type "help", "copyright", "credits" or "license" for more information.

import genesis as gs
gs.init(backend=gs.gpu) # or gs.cuda if CUDA is available
[Genesis] [15:03:55] [INFO] ╭───────────────────────────────────────╮
[Genesis] [15:03:55] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [15:03:55] [INFO] ╰───────────────────────────────────────╯
[Genesis] [15:03:56] [INFO] Running on [NVIDIA GeForce RTX 3050 Ti Laptop GPU] with backend gs.cuda. Device memory: 4.00 GB.
[Genesis] [15:03:56] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
scene = gs.Scene(show_viewer=True)
[Genesis] [15:04:06] [INFO] Scene <8dc41ec> created.
plane = scene.add_entity(gs.morphs.Plane())
[Genesis] [15:04:06] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <73e8b45>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
scene.build()
[Genesis] [15:04:06] [INFO] Building scene <8dc41ec>...
[Genesis] [15:04:30] [INFO] Compiling simulation kernels...
[Genesis] [15:05:44] [INFO] Building visualizer...
Traceback (most recent call last):
File "", line 1, in
File "C:\Users\vicki\anaconda3\envs\genesis_env\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
return method(self, *args, **kwargs)
File "C:\Users\vicki\anaconda3\envs\genesis_env\lib\site-packages\genesis\engine\scene.py", line 556, in build
self._visualizer.build()
File "C:\Users\vicki\anaconda3\envs\genesis_env\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
self._context.build(self._scene)
File "C:\Users\vicki\anaconda3\envs\genesis_env\lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
self.jit = JITRenderer(self._scene, [], [])
File "C:\Users\vicki\anaconda3\envs\genesis_env\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in init
self.gen_func_ptr()
File "C:\Users\vicki\anaconda3\envs\genesis_env\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
self.gl = GLWrapper()
File "C:\Users\vicki\anaconda3\envs\genesis_env\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in init
load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
File "C:\Users\vicki\anaconda3\envs\genesis_env\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
func_ptr = GL.platform.ctypesloader.buildFunction(
File "C:\Users\vicki\anaconda3\envs\genesis_env\lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
return functionType( (name, dll), )
AttributeError: function 'glGetUniformLocation' not found
for i in range(100):
... scene.step()"

aidinism, 2024-12-26:
add this import right before the genesis import:

```
import os
os.environ['PYOPENGL_PLATFORM'] = 'glx'
```

vstavros02, 2024-12-26:
Thank you for the suggestions! However, I tried that and the visualizer still did not work, here was what my terminal output is on WSL virtual environment:

"__(genesis_project) vickistavros@Timmy:~$ python3
Python 3.11.11 (main, Dec 11 2024, 16:28:39) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.environ['PYOPENGL_PLATFORM'] = 'glx'
>>> import genesis as gs
ackend=gs.cpu)

>>> gs.init(backend=gs.cpu)
[Genesis] [18:30:06] [INFO] ╭───────────────────────────────────────╮
[Genesis] [18:30:06] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [18:30:06] [INFO] ╰───────────────────────────────────────╯
[Genesis] [18:30:06] [INFO] Running on [12th Gen Intel(R) Core(TM) i9-12900HK] with backend gs.cpu. Device memory: 7.76 GB.
[Genesis] [18:30:06] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
>>>
>>> scene = gs.Scene(
...     sim_options=gs.options.SimOptions(
...         dt=0.01,
...         gravity=(0, 0, -10.0),
...     ),
...     show_viewer=True,
...     viewer_options=gs.options.ViewerOptions(
...         camera_pos=(3.5, 0.0, 2.5),
...         camera_lookat=(0.0, 0.0, 0.5),
...         camera_fov=40,
...     ),
... )
[Genesis] [18:30:26] [INFO] Scene <fa73b78> created.
>>> plane = scene.add_entity(gs.morphs.Plane())
ka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika[Genesis] [18:30:31] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <66a9e45>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
_panda/panda.xml'),
)>>> franka = scene.add_entity(
...     gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
... )
[Genesis] [18:30:32] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <1951be1>, morph: <gs.morphs.MJCF(file='/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
>>> scene.build()
[Genesis] [18:30:41] [INFO] Building scene <fa73b78>...
r i in range(1000):
    scene.step()
[Genesis] [18:30:52] [INFO] Compiling simulation kernels...
[Genesis] [18:31:24] [INFO] Building visualizer...
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 1126, in _init_and_start_app
    super(Viewer, self).__init__(
  File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/pyglet/window/xlib/__init__.py", line 168, in __init__
    super().__init__(*args, **kwargs)
  File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/pyglet/window/__init__.py", line 527, in __init__
    context = config.create_context(gl.current_context)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/pyglet/gl/xlib.py", line 114, in create_context
    return XlibContext(self, share)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vickistavros/miniconda3/envs/genesis_project/lib/python3.11/site-packages/pyglet/gl/xlib.py", line 149, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context"

SomeSlave, 2024-12-27:
Hi, @vstavros02!
Unfortunately, currently, it is not possible to render on Windows 11.
Please refer to the links below for more information.

[AttributeError: function 'glGetUniformLocation' not found #64](https://github.com/Genesis-Embodied-AI/Genesis/issues/64)
[Installation](https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html)

vstavros02, 2024-12-27:
Thank you for the response! I thought I was the problem here lol

vstavros02, 2024-12-27:
So this wont work even by using WSL?

SomeSlave, 2024-12-27:
> So this wont work even by using WSL?

At its core, WSL is a service virtual machine that emulates the Linux kernel. As far as I remember, some functions don't work as they should, and there are limitations. Unfortunately, I will not say more, because I do not have sufficient knowledge in this area. Do you have the option to use Docker? Or the ability to run on a full-fledged Linux OS?

vstavros02, 2024-12-28:
I'm thinking of converting my Windows to Linux in order to have the visualizations :) What system have you been working on? And do you have any tips for me? Thanks for all of your help, you've saved me a lot of time!

zhouxian, 2024-12-28:
> I'm thinking of converting my Windows to Linux in order to have the visualizations :) What system have you been working on? And do you have any tips for me? Thanks for all of your help, you've saved me a lot of time!

I think any Ubuntu newer than 18.04 would work fine
As a reference the machine we used to develop genesis runs 20.04

vstavros02, 2024-12-29:
That's great! I see that some people are saying certain python versions work with genesis, where as others don't. Which python version do you recommend? Also are you using Pip or Anaconda? And last question haha, are you using a virtual environment to do this all?

ai-bits, 2024-12-29:
Midnight approaching I just skimmed your thread while I'm installing on WSL Ubuntu 24.04 on Win 11 and 2x RTX 4000 Ada w/ Anaconda env Python 3.12.
Got Linux Chrome, Filer and editor installed on Ubuntu that show as apps on Win. No idea if the renderer will get the corner, but we'll see in a moment.
Well, "critical job detector", sth went wrong with the Anaconda genesis env on WSL Ubuntu.

Why don't you spin up a VMware or so Ubuntu and just use CPU if you cannot virtualize your GPU?

Until tomorrow
G.

vstavros02, 2024-12-30:
Thank you for the response @ai-bits! I was really struggling with this issue until @SomeSlave mentioned that the visualizer is not yet compatible with Windows nor WSL. I assumed that it would have worked with WSL but unfortunately everything except for the visualizer works. Since then I have created a Dual Boot system with Linux and Windows by following this tutorial for my specific Windows type "https://youtu.be/uqZIp4ay-3s?si=y4bDLtrvZjRkQURH". After successfully creating the Linux dual system, and installing many prerequisites and those needed to run genesis as shown in the GitHub conversation  "https://github.com/Genesis-Embodied-AI/Genesis/issues/207" commented by  "bradmcgo", I am now successfully able to run the visualizer for Genesis! It did take the full day as I ran into some issues along the way when creating the dual system but most issues came from installing CUDA on my system (I followed tutorials as well but ended up doing what @bradmcgo did, what a life savor he was lol). I'm not sure if you are struggling with the same issue, but if you are and care for the  the visualizing aspect of Genesis, I recommend doing what I did but with caution, and take your time to avoid errors along the way!  And if you do proceed with that, please feel free to message me if you run into any sort of issue or need guidance, mind you I'm fairly new to all of this and with trouble and error, and help from Chatgpt and Claud.ai, as well as the github conversations, I was able to succeed! Anyways I hope I was as helpful as you were, good luck and thanks again!

ai-bits, 2024-12-30:
Glad you found a solution!
A dual boot is way better than a VM, as the popular virtualizers can't virtualize GPUs, but it woulda been the quickest kludge.
WSL2 = WSLg (graphics), really handy in Win for some tasks, CAN run GUI apps (as said I have Chrome, the Filer and the editor as Win apps), but Genesis is very special and the viewer apparently has its problems with WSL's X11 or Wayland or whatever they are using nowadays.
I'm not sure if their Docker solution does really work. It may simplify installation, but it adds a level of complexity in execution.

If you really mean installing CUDA and not the Nvidia driver:
It's years ago that you needed a separate CUDA installation, except for compiling it into your code. You don't need it for Genesis as it is compiled into PyTorch, etc. and that saves you endless version headaches.

Cheers
G.

closed 2025-02-23 by YilingQiao