Genesis-Embodied-AI/genesis-world #345: Runtime Error Validation Failed OpenGL shaders

yang787, 2024-12-27:
Hello, I got the following error, ran the render_on_macos.py with Apple M3 Pro

  File "/Genesis/examples/render_on_macos.py", line 46, in main
    scene.viewer.start()
  File "/.venv/lib/python3.10/site-packages/genesis/vis/viewer.py", line 91, in start
    self._pyrender_viewer.start()
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 386, in start
    self._init_and_start_app()
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/.venv/lib/python3.10/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/.venv/lib/python3.10/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/.venv/lib/python3.10/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 168, in render
    retval = self._forward_pass(scene, flags, seg_node_map=seg_node_map)
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 534, in _forward_pass
    self.jit.forward_pass(
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/jit_render.py", line 708, in forward_pass
    self.load_programs(renderer, flags, program_flags)
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/jit_render.py", line 349, in load_programs
    program = renderer._get_primitive_program(primitive, flags, program_flags)
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 1142, in _get_primitive_program
    program._add_to_context()
  File "/.venv/lib/python3.10/site-packages/genesis/ext/pyrender/shader_program.py", line 125, in _add_to_context
    self._program_id = gl_shader_utils.compileProgram(*shader_ids, validate=False)
  File "/.venv/lib/python3.10/site-packages/OpenGL/GL/shaders.py", line 196, in compileProgram
    program.check_validate()
  File "/.venv/lib/python3.10/site-packages/OpenGL/GL/shaders.py", line 105, in check_validate
    raise RuntimeError(
RuntimeError: Validation failure (0): b"Validation Failed: Sampler error:\n  Samplers of different types use the same texture image unit.\n   - or -\n  A sampler's texture unit is out of range (greater than max allowed or negative).\n"

iCatchYou, 2025-01-10:
Hello, I have encountered the same problem. How did you solve it？

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 1161, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 955, in _time_event
    self.on_draw()
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._render()
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 1098, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/renderer.py", line 168, in render
    retval = self._forward_pass(scene, flags, seg_node_map=seg_node_map)
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/renderer.py", line 534, in _forward_pass
    self.jit.forward_pass(
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/jit_render.py", line 708, in forward_pass
    self.load_programs(renderer, flags, program_flags)
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/jit_render.py", line 349, in load_programs
    program = renderer._get_primitive_program(primitive, flags, program_flags)
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/renderer.py", line 1142, in _get_primitive_program
    program._add_to_context()
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/ext/pyrender/shader_program.py", line 125, in _add_to_context
    self._program_id = gl_shader_utils.compileProgram(*shader_ids, validate=False)
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/OpenGL/GL/shaders.py", line 196, in compileProgram
    program.check_validate()
  File "/home/yangze/miniconda3/envs/torch/lib/python3.9/site-packages/OpenGL/GL/shaders.py", line 105, in check_validate
    raise RuntimeError(
RuntimeError: Validation failure (0): b'active samplers with a different type refer to the same texture image unit'

duburcqa, 2025-02-27:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/pull/777