Genesis-Embodied-AI/genesis-world #352: I am using Windows WSL2 with Ubuntu 22.04, and I encountered this error. Could you please help me?

faridhossainrd, 2024-12-27:
[Genesis] [15:03:36] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [15:03:36] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [15:03:36] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [15:03:36] [INFO] Running on [NVIDIA GeForce GTX 1060] with backend gs.cuda. Device memory: 6.00 GB.
[W 12/27/24 15:03:36.649 113286] [cuda_driver.cpp:load_lib@36] libcuda.so lib not found.
[W 12/27/24 15:03:36.650 113286] [misc.py:adaptive_arch_select@758] Arch=[<Arch.cuda: 3>] is not supported, falling back to CPU
[Genesis] [15:03:36] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [15:03:37] [INFO] Scene <c872814> created.
[Genesis] [15:03:37] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <61039fd>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [15:03:37] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <14491b7>, morph: <gs.morphs.MJCF(file='/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [15:03:38] [INFO] Building scene <c872814>...
[Genesis] [15:03:44] [INFO] Compiling simulation kernels...
[Genesis] [15:03:51] [INFO] Building visualizer...
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/genesis/ext/pyrender/renderer.py", line 899, in _update_context
    p._add_to_context()
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/genesis/ext/pyrender/primitive.py", line 359, in _add_to_context
    glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, FLOAT_SZ * 3, ctypes.c_void_p(0))
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/OpenGL/latebind.py", line 63, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/OpenGL/GL/VERSION/GL_2_0.py", line 469, in glVertexAttribPointer
    contextdata.setValue( key, array )
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
  File "/home/farid/anaconda3/envs/robots/lib/python3.9/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context

Kashu7100, 2024-12-27:
It seems you have several problems. First, check your libcuda.so exists (or added to path). Also try to see if you have `libglib2.0-0`, `libegl1`, `libgles2` on your machine.

zhangiser, 2024-12-28:
import os before import genesis,see [https://blog.csdn.net/xiangshangdemayi/article/details/144728092](url)
eg：
```python
import os
os.environ['PYOPENGL_PLATFORM'] = 'glx'
import genesis as gs

gs.init(backend=gs.cuda)

scene = gs.Scene()
```

faridhossainrd, 2024-12-28:
Thank you so much it's solved but got a Ray Tracing Rendering build error, could you please help with it?

cmake --build build -j $(nproc)
[  0%] Building CXX object src/ext/CMakeFiles/tinyexr.dir/tinyexr.cpp.o
[  0%] Built target luisa-compute-include
[  0%] Building C object src/compute/src/ext/stb/CMakeFiles/stb.dir/stb.c.o
[  0%] Building C object src/compute/src/ext/CMakeFiles/luisa-compute-ext-lmdb.dir/liblmdb/mdb.c.o
[  0%] Built target luisa-compute-ext-copy
[  0%] Building C object src/compute/src/ext/glfw/src/CMakeFiles/glfw.dir/context.c.o
[  0%] Building CXX object src/compute/src/ext/EASTL/CMakeFiles/EASTL.dir/source/allocator_luisa.cpp.o
[  0%] Building CXX object src/compute/src/ext/EASTL/CMakeFiles/EASTL.dir/source/allocator_eastl.cpp.o
/home/farid/Genesis/genesis/ext/LuisaRender/src/ext/tinyexr.cpp:8:10: fatal error: zlib.h: No such file or directory
    8 | #include <zlib.h>
      |          ^~~~~~~~
compilation terminated.
gmake[2]: *** [src/ext/CMakeFiles/tinyexr.dir/build.make:76: src/ext/CMakeFiles/tinyexr.dir/tinyexr.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1265: src/ext/CMakeFiles/tinyexr.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
[  0%] Building CXX object src/ext/assimp/code/CMakeFiles/assimp.dir/Common/Assimp.cpp.o
[  0%] Building C object src/compute/src/ext/glfw/src/CMakeFiles/glfw.dir/init.c.o
[  0%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/clock.posix.c.o
In file included from /home/farid/Genesis/genesis/ext/LuisaRender/src/compute/src/ext/glfw/src/platform.h:81,
                 from /home/farid/Genesis/genesis/ext/LuisaRender/src/compute/src/ext/glfw/src/internal.h:334,
                 from /home/farid/Genesis/genesis/ext/LuisaRender/src/compute/src/ext/glfw/src/context.c:28:
/home/farid/Genesis/genesis/ext/LuisaRender/src/compute/src/ext/glfw/src/x11_platform.h:32:10: fatal error: X11/Xlib.h: No such file or directory
   32 | #include <X11/Xlib.h>
      |          ^~~~~~~~~~~~
compilation terminated.
In file included from /home/farid/Genesis/genesis/ext/LuisaRender/src/compute/src/ext/glfw/src/platform.h:81,
                 from /home/farid/Genesis/genesis/ext/LuisaRender/src/compute/src/ext/glfw/src/internal.h:334,
                 from /home/farid/Genesis/genesis/ext/LuisaRender/src/compute/src/ext/glfw/src/init.c:28:
/home/farid/Genesis/genesis/ext/LuisaRender/src/compute/src/ext/glfw/src/x11_platform.h:32:10: fatal error: X11/Xlib.h: No such file or directory
   32 | #include <X11/Xlib.h>
      |          ^~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [src/compute/src/ext/glfw/src/CMakeFiles/glfw.dir/build.make:76: src/compute/src/ext/glfw/src/CMakeFiles/glfw.dir/context.c.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [src/compute/src/ext/glfw/src/CMakeFiles/glfw.dir/build.make:90: src/compute/src/ext/glfw/src/CMakeFiles/glfw.dir/init.c.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1525: src/compute/src/ext/glfw/src/CMakeFiles/glfw.dir/all] Error 2
[  0%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/drain.c.o
[  0%] Building CXX object src/ext/assimp/code/CMakeFiles/assimp.dir/CApi/CInterfaceIOWrapper.cpp.o
[  0%] Building C object src/compute/src/ext/CMakeFiles/luisa-compute-ext-lmdb.dir/liblmdb/midl.c.o
[  0%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/error.posix.c.o
[  0%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/handle.posix.c.o
[  0%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/init.posix.c.o
[  0%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/options.c.o
[  0%] Building CXX object src/compute/src/ext/EASTL/CMakeFiles/EASTL.dir/source/assert.cpp.o
[  0%] Building CXX object src/compute/src/ext/EASTL/CMakeFiles/EASTL.dir/source/atomic.cpp.o
[  1%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/process.posix.c.o
[  1%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/pipe.posix.c.o
[  1%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/redirect.posix.c.o
[  1%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/redirect.c.o
[  1%] Building CXX object src/compute/src/ext/EASTL/CMakeFiles/EASTL.dir/source/fixed_pool.cpp.o
[  1%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/reproc.c.o
[  1%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/run.c.o
[  1%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/strv.c.o
[  1%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/utf.posix.c.o
[  1%] Building CXX object src/ext/assimp/code/CMakeFiles/assimp.dir/Common/Compression.cpp.o
[  2%] Building CXX object src/ext/assimp/code/CMakeFiles/assimp.dir/Common/BaseImporter.cpp.o
[  2%] Built target reproc
[  2%] Building CXX object src/compute/src/ext/EASTL/CMakeFiles/EASTL.dir/source/hashtable.cpp.o
[  2%] Building CXX object src/compute/src/ext/EASTL/CMakeFiles/EASTL.dir/source/intrusive_list.cpp.o
In file included from /home/farid/Genesis/genesis/ext/LuisaRender/src/ext/assimp/code/Common/Compression.cpp:42:
/home/farid/Genesis/genesis/ext/LuisaRender/src/ext/assimp/code/Common/Compression.h:44:10: fatal error: zlib.h: No such file or directory
   44 | #include "zlib.h"
      |          ^~~~~~~~
compilation terminated.
gmake[2]: *** [src/ext/assimp/code/CMakeFiles/assimp.dir/build.make:104: src/ext/assimp/code/CMakeFiles/assimp.dir/Common/Compression.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....

Kashu7100, 2024-12-28:
Have you recursively init the submodules?

closed 2025-02-28 by duburcqa

Cylrx, 2025-08-19:
> Thank you so much it's solved but got a Ray Tracing Rendering build error, could you please help with it?
>
> ```
> cmake --build build -j $(nproc)
> [  0%] Building CXX object [...]

Hi, I ran into the same error with "zlib.h" when building `LuisaRender`. And, yes, I have recursively init the submodules.