Genesis-Embodied-AI/genesis-world #353: win11 wsl ubuntu CUDA_ERROR_NO_DEVICE

zhangiser, 2024-12-27:
I'm used win11 wsl ubuntu,run Hello Genesis got Error: CUDA_ERROR_NO_DEVICE:
```txt
(torch) root@zhangiser-pc:/usr/Genesis-main# /root/miniconda3/envs/torch/bin/python /usr/Genesis-main/examples/tutorials/hello_genesis.py
[Genesis] [09:35:05] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [09:35:05] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [09:35:05] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [09:35:06] [INFO] Running on [NVIDIA GeForce RTX 4080 Laptop GPU] with backend gs.cuda. Device memory: 11.99 GB.
[E 12/27/24 09:35:07.263 1371] [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected while calling init (cuInit)

Traceback (most recent call last):
  File "/usr/Genesis-main/examples/tutorials/hello_genesis.py", line 3, in <module>
    gs.init(backend=gs.cuda)
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/__init__.py", line 97, in init
    ti.init(arch=TI_ARCH[platform][backend], debug=debug, force_scalarize_matrix=True)
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/taichi/lang/misc.py", line 458, in init
    impl.get_runtime().create_program()
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/taichi/lang/impl.py", line 388, in create_program
    self.prog = _ti_core.Program()
RuntimeError: [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected while calling init (cuInit)
[Genesis] [09:35:07] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

wsl info:
```txt
(base) root@zhangiser-pc:/usr/Genesis-main# wsl -l -v
  NAME                   STATE           VERSION
* docker-desktop-data    Stopped         2
  docker-desktop         Stopped         2
  Ubuntu-22.04           Running         2
```
CUDA info:
```txt
(base) root@zhangiser-pc:/usr/Genesis-main# nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2024 NVIDIA Corporation
Built on Thu_Mar_28_02:18:24_PDT_2024
Cuda compilation tools, release 12.4, V12.4.131
Build cuda_12.4.r12.4/compiler.34097967_0
```
nvidia-smi info:
```txt
(base) root@zhangiser-pc:/usr/Genesis-main# nvidia-smi
Fri Dec 27 17:07:40 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 565.77.01              Driver Version: 566.36         CUDA Version: 12.7     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4080 ...    On  |   00000000:01:00.0 Off |                  N/A |
| N/A   48C    P0            593W /  155W |       0MiB /  12282MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```
Pytorch info:
```txt
(base) root@zhangiser-pc:/usr/Genesis-main# conda activate torch
(torch) root@zhangiser-pc:/usr/Genesis-main# python
Python 3.9.21 (main, Dec 11 2024, 16:24:11) 
[GCC 11.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> torch.cuda.is_available()
True
```
Everying seems OK! But why got CUDA_ERROR_NO_DEVICE ? Is there anyone know this problem?

a3ahmad, 2024-12-29:
This worked for me:

https://github.com/Genesis-Embodied-AI/Genesis/issues/187#issuecomment-2558133960

zhangiser, 2024-12-30:
> This worked for me:
> 
> [#187 (comment)](https://github.com/Genesis-Embodied-AI/Genesis/issues/187#issuecomment-2558133960)

That says do something with env
```bash
export LD_LIBRARY_PATH=/usr/lib/wsl/lib:$LD_LIBRARY_PATH
source ~/.bashrc
```
Actually,I set it up a long time ago.So that isn't worked for me.
I have tried reinstall CUDA so many times,but the error always there.
Finally, I reinstalled the Ubuntu feeling helpless.

closed 2024-12-30 by zhangiser