Genesis-Embodied-AI/genesis-world #354: docs Python -V must be >=3.9 AND <3.13 and install PyTorch before Genesis fixes numpy ver problems

ai-bits, 2024-12-27:
As even Hello ran into issues with the suggested setup (mainly numpy, e.g. ValueError) on any platform (x64 Ubuntu 24.04 & Win 11, MBP M3) I put some more work and analysis.

First off, install runs into all kinds of issues with Python 3.13, so docs should say Python >=3.9 **<3.13**.

Running into numpy errors I installed PyTorch **before** Genesis (from latest repo) in a 3.12(.8) env, which replaced numpy 2.0.1 with 1.26.4 from Genesis which made Hello run.

Cheers
G.

zhouxian, 2024-12-27:
what's causing issue with 3.13?

ai-bits, 2024-12-27:
Under the summary is the output from my M3 MBP when I tried to install Genesis and PyTorch latest in a clean Conda Python 3.13.1 env.
Problems with:
1) taichi==1.7.2
2) torchvision

Also, as said, to avoid numpy errors from PyTorch numpy 2.0.1 and make Genesis work AT ALL, install PyTorch first and THEN Genesis with its numpy 1.26.4, replacing 2.0.1 (and pillow 11.0.0?). (seemingly incompatibility between numpy v 1 & 2 and Genesis needs v 1)
HTH
G.

(base) max:Genesis gy$ conda activate genesis
(genesis) max:Genesis gy$ pip install -e .
Obtaining file:///Users/gy/dl/Genesis
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting psutil (from genesis-world==0.2.0)
  Downloading psutil-6.1.1-cp36-abi3-macosx_11_0_arm64.whl.metadata (22 kB)
Collecting scikit-image (from genesis-world==0.2.0)
  Downloading scikit_image-0.25.0-cp313-cp313-macosx_12_0_arm64.whl.metadata (14 kB)
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement taichi==1.7.2 (from genesis-world) (from versions: none)
ERROR: No matching distribution found for taichi==1.7.2
(genesis) max:Genesis gy$ pip install genesis-world
Collecting genesis-world
  Downloading genesis_world-0.2.0-py3-none-any.whl.metadata (11 kB)
Collecting psutil (from genesis-world)
  Using cached psutil-6.1.1-cp36-abi3-macosx_11_0_arm64.whl.metadata (22 kB)
Collecting scikit-image (from genesis-world)
  Using cached scikit_image-0.25.0-cp313-cp313-macosx_12_0_arm64.whl.metadata (14 kB)
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
Collecting genesis-world
  Downloading genesis_world-0.1.1-py3-none-any.whl.metadata (11 kB)
ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.0 depends on taichi==1.7.2
    genesis-world 0.1.1 depends on taichi==1.7.2

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
(genesis) max:Genesis gy$ conda install pytorch::pytorch torchvision torchaudio -c pytorch
Channels:
 - pytorch
 - defaults
Platform: osx-arm64
Collecting package metadata (repodata.json): done
Solving environment: | warning  libmamba Added empty dependency for problem type SOLVER_RULE_UPDATE
failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package torchvision-0.13.0-py310_cpu requires python >=3.10,<3.11.0a0, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ pin-1 is installable and it requires
│  └─ python 3.13.* , which can be installed;
└─ torchvision is not installable because there are no viable options
   ├─ torchvision [0.11.3|0.13.0|...|0.20.1] would require
   │  └─ python >=3.10,<3.11.0a0 , which conflicts with any installable versions previously reported;
   ├─ torchvision [0.11.3|0.13.0|...|0.20.0] would require
   │  └─ python >=3.8,<3.9.0a0 , which conflicts with any installable versions previously reported;
   ├─ torchvision [0.11.3|0.13.0|...|0.20.1] would require
   │  └─ python >=3.9,<3.10.0a0 , which conflicts with any installable versions previously reported;
   ├─ torchvision [0.15.2|0.16.0|...|0.20.1] would require
   │  └─ python >=3.11,<3.12.0a0 , which conflicts with any installable versions previously reported;
   └─ torchvision [0.17.0|0.17.1|...|0.20.1] would require
      └─ python >=3.12,<3.13.0a0 , which conflicts with any installable versions previously reported.

zhouxian, 2024-12-29:
in the pyproject.toml file we specified numpy=1.26 -- numpy 2 has various compatibility issues with many packages

ai-bits, 2024-12-29:
Sure, and because PyTorch (at least latest) overwrites Genesis' numpy 1.26 when you follow the docs, they should be changed to install PyTorch FIRST and THEN Genesis to overwrite 2.0 with 1.26 and not vv.

And because torchvisoin,.. has problems with 3.13, docs should say python<3.13.
And because - as stated in the "minimum Python version.." thread - pymeshlab and macOS 14 have problems with 3.9, docs should recommend python >=3.10 and <3.13.

These issues are not apparent when you adapted your dev environment over time, but in a clean install.

Thanks
G.

zhouxian, 2024-12-29:
Got it! thanks for the heads up. Will update doc

duburcqa, 2025-03-23:
Documentation has been updated. It will be reflected at next release.

closed 2025-03-23 by duburcqa