Genesis-Embodied-AI/genesis-world #361: Fatal Error: unable to install Genesis

younes-io, 2024-12-27:
Unfortunately, I have several times for a couple of hours now, by executing the command as mentioned in the README `docker build -t genesis -f docker/Dockerfile docker` and I eventually get this:

[image]

I'm on WSL2 Debian, and I have dedicated 16GB of RAM to WSL2

younes-io, 2024-12-28:
@zhouxian : could you please help with this?

zhouxian, 2024-12-28:
@Kashu7100

younes-io, 2024-12-28:
I just realized the issue might be related to the fact I'm running this on Debian and in the dockerfile I can see `RUN add-apt-repository ppa:ubuntu-toolchain-r/test && \`

Kashu7100, 2024-12-28:
Yeah that can be a problem. Also, if you don't need the rendering capability, you can comment out the Luisa build part, which will simplify the entire pipeline (will be a good sanity check).

younes-io, 2024-12-28:
thanks @Kashu7100 , I actually need the rendering capability.. I'm installing Ubuntu 24 now, and I'll try this out and see.. I'll close this issue though as it's related to my Debian distro

closed 2024-12-28 by younes-io