Genesis-Embodied-AI/genesis-world #364: How to install LuisaCompute manually?

zhangiser, 2024-12-28:
I'm used windows wsl ubuntu, I just download genesis project as a zip file,and copy it into ubuntu path.
Now i'm running examples/rendering/demo.py,and got ModuleNotFoundError: No module named 'LuisaRenderPy'.
[image]
In office doc install context, i found the this
```bash
git submodule update --init --recursive
```
But i didn't use git, so i can't install LuisaCompute by this command.
Can i also download [LuisaCompute](https://github.com/LuisaGroup/LuisaCompute) as a zip file and put it on genesis/ext/LuisaRender,
then install it by
```bash
cd genesis/ext/LuisaRender
cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF 
cmake --build build -j $(nproc)
```

zhouxian, 2024-12-28:
Yes sure you can download it separately and compile
make sure you download the correct commit

zhangiser, 2024-12-28:
> Yes sure you can download it separately and compile make sure you download the correct commit

Thanks,But I don't know which commit is correct,can you suggest some commit?

zhouxian, 2024-12-28:
see : https://github.com/Genesis-Embodied-AI/Genesis/tree/main/genesis/ext

closed 2024-12-28 by zhouxian