Genesis-Embodied-AI/genesis-world #366: ompl的solved = ss.solve(timeout)，Segmentation fault (core dumped)

zhaianbang, 2024-12-28:
(genesis) ascend@tchen-Raider-GE76-12UHS:/media/ascend/diskb/zab2025/genesis_code/Genesis$ python3 /media/ascend/diskb/zab2025/genesis_code/Genesis/examples/rigid/grasp_bottle.py
[Genesis] [16:09:36] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [16:09:36] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [16:09:36] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [16:09:36] [INFO] Running on [NVIDIA GeForce RTX 3080 Ti Laptop GPU] with backend gs.cuda. Device memory: 15.74 GB.
[Genesis] [16:09:36] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [16:09:37] [INFO] Scene <1e4db73> created.
[Genesis] [16:09:37] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <0a5b4b2>, morph: <gs.morphs.URDF(file='/media/ascend/diskb/zab2025/genesis_code/Genesis/genesis/assets/urdf/plane/plane.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [16:09:37] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <a4b926e>, morph: <gs.morphs.URDF(file='/media/ascend/diskb/zab2025/genesis_code/Genesis/genesis/assets/urdf/3763/mobility_vhacd.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [16:09:37] [INFO] Adding <gs.RigidEntity>. idx: 2, uid: <56b04b5>, morph: <gs.morphs.MJCF(file='/media/ascend/diskb/zab2025/genesis_code/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [16:09:37] [INFO] Building scene <1e4db73>...
[Genesis] [16:09:41] [INFO] Compiling simulation kernels...
[Genesis] [16:09:45] [INFO] Building visualizer...
Computed qpos from IK: tensor([-1.0432,  1.4372,  1.5254, -1.7213, -1.4453,  1.6352,  1.4565,  0.0112,
         0.0096], device='cuda:0')
Joint limits: (tensor([-2.8973, -1.7628, -2.8973, -3.0718, -2.8973, -0.0175, -2.8973,  0.0000,
         0.0000], device='cuda:0'), tensor([ 2.8973,  1.7628,  2.8973, -0.0698,  2.8973,  3.7525,  2.8973,  0.0400,
         0.0400], device='cuda:0'))
qpos_start =  [0. 0. 0. 0. 0. 0. 0. 0. 0.]
q_limit_lower =  [-2.8973 -1.7628 -2.8973 -3.0718 -2.8973 -0.0175 -2.8973  0.      0.    ]  q_limit_upper =  [2.8973 1.7628 2.8973 3.     2.8973 3.7525 2.8973 0.04   0.04  ]
1312
1313
Segmentation fault (core dumped)

在运行python3 /media/ascend/diskb/zab2025/genesis_code/Genesis/examples/rigid/grasp_bottle.py的时候，运行到genesis_code/Genesis/genesis/engine/entities/rigid_entity/rigid_entity.py中的solved = ss.solve(timeout)时，出现段错误，没有使用docker，用的conda环境，怎么解决呢？

zhouxian, 2024-12-28:
这个应该是ompl的问题。。。可能得在ompl上提个issue
用ompl是一个临时的解决方案，我们尽快放出gpu加速的自己手搓的rrt

zhaianbang, 2025-01-06:
```
ompl的segment fault的问题，解决方案，已解决：
运行genesis，直接ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl，运行grasp_bottle程序出现ompl的segment fault的问题，解决步骤如下：
（重点：注意查看/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/lib中没有编译出来的4个包的报错log（_base.so _control.so _geometric.so _util.so），然后对应去解决）
用python=3.10的conda环境：
conda create genesis0106 python=3.10

conda activate enesis0106

pip install torch==2.5.1 torchvision torchaudio --index-url https://download.pytorch.org/whl/cu118

sudo apt-get -y install g++ cmake pkg-config libboost-serialization-dev libboost-filesystem-dev libboost-system-dev libboost-program-options-dev libboost-test-dev libeigen3-dev libode-dev wget libyaml-cpp-dev
export CXX=g++
export MAKEFLAGS="-j `nproc`"
pip install -vU https://github.com/CastXML/pygccxml/archive/develop.zip pyplusplus
sudo apt-get -y install castxml libboost-python-dev libboost-numpy-dev pypy3

git clone --recurse-submodules https://github.com/ompl/ompl.git -b 1.6.0
mkdir -p ompl/build
cd ompl/build

cmake .. -DPYTHON_EXEC=$(which python) -DCMAKE_INSTALL_PREFIX="$CONDA_PREFIX" -DBOOST_ROOT="$CONDA_PREFIX" -DPYTHONPATH=$CONDA_PREFIX/lib/python3.10/site-packages

注意查看/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/lib中没有编译出来的4个包的报错log（_base.so _control.so _geometric.so _util.so），然后对应去解决：
/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/pyplusplus_util.log

如果出现cannot import name 'Protocol' from 'typing'错误：
找到对应的行，然后注释掉typing，替换加入from typing_extensions import Protocol
pip install typing_extensions==4.5.0
pip install pygccxml==2.1.0

修改注释/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/py-bindings/generate_bindings.py中194行的：

    # self.ompl_ns.class_(f'SpecificParam< {self.string_decl} >').rename('SpecificParamString')
    try:
        self.ompl_ns.class_(f'SpecificParam< std::string >').rename('SpecificParamString')
    except:
        self.ompl_ns.class_(f'SpecificParam< std::basic_string< char > >').rename('SpecificParamString')

然后执行：
make update_bindings
（如果出现No rule for update_bindings报错，执行conda install -c conda-forge boost boost-cpp）

再执行：
make -j8
sudo make install

欢迎交流ab_zhai@foxmail.com
```

zhaianbang, 2025-01-06:
> 这个应该是ompl的问题。。。可能得在ompl上提个issue 用ompl是一个临时的解决方案，我们尽快放出gpu加速的自己手搓的rrt

```
ompl的segment fault的问题，解决方案，已解决：
运行genesis，直接ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl，运行grasp_bottle程序出现ompl的segment fault的问题，解决步骤如下：
（重点：注意查看/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/lib中没有编译出来的4个包的报错log（_base.so _control.so _geometric.so _util.so），然后对应去解决）
用python=3.10的conda环境：
conda create genesis0106 python=3.10

conda activate enesis0106

pip install torch==2.5.1 torchvision torchaudio --index-url https://download.pytorch.org/whl/cu118

sudo apt-get -y install g++ cmake pkg-config libboost-serialization-dev libboost-filesystem-dev libboost-system-dev libboost-program-options-dev libboost-test-dev libeigen3-dev libode-dev wget libyaml-cpp-dev
export CXX=g++
export MAKEFLAGS="-j `nproc`"
pip install -vU https://github.com/CastXML/pygccxml/archive/develop.zip pyplusplus
sudo apt-get -y install castxml libboost-python-dev libboost-numpy-dev pypy3

git clone --recurse-submodules https://github.com/ompl/ompl.git -b 1.6.0
mkdir -p ompl/build
cd ompl/build

cmake .. -DPYTHON_EXEC=$(which python) -DCMAKE_INSTALL_PREFIX="$CONDA_PREFIX" -DBOOST_ROOT="$CONDA_PREFIX" -DPYTHONPATH=$CONDA_PREFIX/lib/python3.10/site-packages

注意查看/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/lib中没有编译出来的4个包的报错log（_base.so _control.so _geometric.so _util.so），然后对应去解决：
/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/pyplusplus_util.log

如果出现cannot import name 'Protocol' from 'typing'错误：
找到对应的行，然后注释掉typing，替换加入from typing_extensions import Protocol
pip install typing_extensions==4.5.0
pip install pygccxml==2.1.0

修改注释/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/py-bindings/generate_bindings.py中194行的：

    # self.ompl_ns.class_(f'SpecificParam< {self.string_decl} >').rename('SpecificParamString')
    try:
        self.ompl_ns.class_(f'SpecificParam< std::string >').rename('SpecificParamString')
    except:
        self.ompl_ns.class_(f'SpecificParam< std::basic_string< char > >').rename('SpecificParamString')

然后执行：
make update_bindings
（如果出现No rule for update_bindings报错，执行conda install -c conda-forge boost boost-cpp）

再执行：
make -j8
sudo make install

欢迎交流ab_zhai@foxmail.com
```

duburcqa, 2025-03-23:
Is this issue fixed or not ?

closed 2025-03-24 by duburcqa