Genesis-Embodied-AI/genesis-world #368: Cannot see graphical view on M3 Macbook

samhincks, 2024-12-28:
Hello, awesome code. I cant wait to get my hands dirty with this 
I run this code on my M3 Mac: 
```

import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()
```

And then I see this output in the terminal 
[image]

However there is no graphical view appearing, so I cannot see the effect of the simulation... 

Perhaps its related to the warning I received at the code initialization
```

% python render_on_macos.py
objc[17890]: Class GLFWHelper is implemented in both /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x143af8570) and /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x178f1d7c0). One of the two will be used. Which one is undefined.
objc[17890]: Class GLFWApplicationDelegate is implemented in both /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x143af85c0) and /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x178f1d810). One of the two will be used. Which one is undefined.
objc[17890]: Class GLFWWindowDelegate is implemented in both /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x143af85e8) and /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x178f1d838). One of the two will be used. Which one is undefined.
objc[17890]: Class GLFWContentView is implemented in both /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x143af8638) and /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x178f1d888). One of the two will be used. Which one is undefined.
objc[17890]: Class GLFWWindow is implemented in both /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x143af86b0) and /opt/anaconda3/envs/genesis/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x178f1d900). 
```

Please help :)

andrewlyu, 2024-12-29:
I ran into the exact same issue on m1 macbook.

Please help +1

Felix-Zhenghao, 2024-12-29:
Same issue on M3 Mac with macOS 14.

```
objc[26828]: Class GLFWHelper is implemented in both /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x159e00570) and /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x3159cd7c0). One of the two will be used. Which one is undefined.
objc[26828]: Class GLFWApplicationDelegate is implemented in both /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x159e005c0) and /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x3159cd810). One of the two will be used. Which one is undefined.
objc[26828]: Class GLFWWindowDelegate is implemented in both /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x159e005e8) and /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x3159cd838). One of the two will be used. Which one is undefined.
objc[26828]: Class GLFWContentView is implemented in both /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x159e00638) and /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x3159cd888). One of the two will be used. Which one is undefined.
objc[26828]: Class GLFWWindow is implemented in both /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x159e006b0) and /opt/miniconda3/envs/genesis/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x3159cd900). One of the two will be used. Which one is undefined.
```

Felix-Zhenghao, 2024-12-29:
@andrewlyu @samhincks 

You can run the script below and you will see render results on your screen.

It is modified from [this script](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/control_your_robot.html) in the official doc. An example provided by the official team is [here](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/render_on_macos.py).

```python
import numpy as np

import genesis as gs

########################## init ##########################
gs.init(backend=gs.metal)

########################## create a scene ##########################
scene = gs.Scene(
    viewer_options = gs.options.ViewerOptions(
        camera_pos    = (0, -3.5, 2.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 30,
        res           = (960, 640),
        max_FPS       = 60,
    ),
    sim_options = gs.options.SimOptions(
        dt = 0.01,
    ),
    show_viewer = True,
)

########################## entities ##########################
plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.MJCF(
        file  = 'xml/franka_emika_panda/panda.xml',
    ),
)
########################## build ##########################
scene.build()

jnt_names = [
    'joint1',
    'joint2',
    'joint3',
    'joint4',
    'joint5',
    'joint6',
    'joint7',
    'finger_joint1',
    'finger_joint2',
]
dofs_idx = [franka.get_joint(name).dof_idx_local for name in jnt_names]

############ Optional: set control gains ############
# set positional gains
franka.set_dofs_kp(
    kp             = np.array([4500, 4500, 3500, 3500, 2000, 2000, 2000, 100, 100]),
    dofs_idx_local = dofs_idx,
)
# set velocity gains
franka.set_dofs_kv(
    kv             = np.array([450, 450, 350, 350, 200, 200, 200, 10, 10]),
    dofs_idx_local = dofs_idx,
)
# set force range for safety
franka.set_dofs_force_range(
    lower          = np.array([-87, -87, -87, -87, -12, -12, -12, -100, -100]),
    upper          = np.array([ 87,  87,  87,  87,  12,  12,  12,  100,  100]),
    dofs_idx_local = dofs_idx,
)

####################################
# package the whole simulation process in a function so that we can launch a thread for it later
####################################
def run_sim(scene,):
    # Hard reset
    for i in range(150):
        if i < 50:
            franka.set_dofs_position(np.array([1, 1, 0, 0, 0, 0, 0, 0.04, 0.04]), dofs_idx)
        elif i < 100:
            franka.set_dofs_position(np.array([-1, 0.8, 1, -2, 1, 0.5, -0.5, 0.04, 0.04]), dofs_idx)
        else:
            franka.set_dofs_position(np.array([0, 0, 0, 0, 0, 0, 0, 0, 0]), dofs_idx)

        scene.step()

    # PD control
    for i in range(1250):
        if i == 0:
            franka.control_dofs_position(
                np.array([1, 1, 0, 0, 0, 0, 0, 0.04, 0.04]),
                dofs_idx,
            )
        elif i == 250:
            franka.control_dofs_position(
                np.array([-1, 0.8, 1, -2, 1, 0.5, -0.5, 0.04, 0.04]),
                dofs_idx,
            )
        elif i == 500:
            franka.control_dofs_position(
                np.array([0, 0, 0, 0, 0, 0, 0, 0, 0]),
                dofs_idx,
            )
        elif i == 750:
            # control first dof with velocity, and the rest with position
            franka.control_dofs_position(
                np.array([0, 0, 0, 0, 0, 0, 0, 0, 0])[1:],
                dofs_idx[1:],
            )
            franka.control_dofs_velocity(
                np.array([1.0, 0, 0, 0, 0, 0, 0, 0, 0])[:1],
                dofs_idx[:1],
            )
        elif i == 1000:
            franka.control_dofs_force(
                np.array([0, 0, 0, 0, 0, 0, 0, 0, 0]),
                dofs_idx,
            )
        # This is the control force computed based on the given control command
        # If using force control, it's the same as the given control command
        print('control force:', franka.get_dofs_control_force(dofs_idx))

        # This is the actual force experienced by the dof
        print('internal force:', franka.get_dofs_force(dofs_idx))

        scene.step()

    scene.viewer.stop()

gs.tools.run_in_another_thread(fn=run_sim, args=(scene,)) # start the simulation in another thread
scene.viewer.start() # start the viewer in the main thread (the render thread)
```

closed 2024-12-29 by ziyanx02

samhincks, 2024-12-29:
Thank you @Felix-Zhenghao 
I ran your code, however it still doesnt work work, giving me this error 
For the record, I am in Python 3.12.8... Are others perhaps using an older version 

```

Traceback (most recent call last):
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/latebind.py", line 41, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/samuelhincks/Documents/Github/genesis/guthib.py", line 126, in <module>
    scene.viewer.start() # start the viewer in the main thread (the render thread)
    ^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/vis/viewer.py", line 91, in start
    self._pyrender_viewer.start()
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 386, in start
    self._init_and_start_app()
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 916, in _update_context
    texture._add_to_context()
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/texture.py", line 180, in _add_to_context
    self._texid = glGenTextures(1)
                  ^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/latebind.py", line 61, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/GL/exceptional.py", line 178, in glGenTextures
    baseFunction( count, textures)
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/latebind.py", line 45, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/wrapper.py", line 657, in wrapperCall
    result = wrappedOperation( *cArguments )
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/platform/baseplatform.py", line 402, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
ctypes.ArgumentError: ("argument 2: TypeError: No array-type handler for type _ctypes.type (value: <cparam 'P' (0x313fe4318)>) registered", (1, <cparam 'P' (0x313fe4318)>))
```

samhincks, 2024-12-29:
I switched to python 3.11, and then the code worked :)

andrewlyu, 2024-12-29:
That script fails for another reason with python 3.11 and macos 15.2. It is a m1 chip macbook.
Following is the traceback report. Looks like a visualization library issue:
```
uniform not found: material.base_color_factor
uniform not found: material.metallic_factor
uniform not found: material.roughness_factor
uniform not found: material.emissive_factor
Traceback (most recent call last):
  File "/Users/duchaolv/projects/genesis_playground/scripts/vis_macos.py", line 126, in <module>
    scene.viewer.start() # start the viewer in the main thread (the render thread)
    ^^^^^^^^^^^^^^^^^^^^
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/genesis/vis/viewer.py", line 91, in start
    self._pyrender_viewer.start()
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 386, in start
    self._init_and_start_app()
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._renderer.render_texts(
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 258, in render_texts
    self._configure_forward_pass_viewport(0)
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 1160, in _configure_forward_pass_viewport
    glBindFramebuffer(GL_DRAW_FRAMEBUFFER, 0)
  File "/Users/duchaolv/projects/brew_envs/genesis_env/lib/python3.11/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glBindFramebuffer,
	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
)

```