Genesis-Embodied-AI/genesis-world #369: The computer directly crashes when training on the 4090.

formoree, 2024-12-28:
I have been using the current computer with a 4090 graphics card to run code based on the IsaacGym framework without any issues. Recently, I migrated the humanoid task to Genesis, and it trained normally on a 4060 Ti. Today, I uploaded the code to the 4090, but every time I attempt to train, the system freezes, and the mouse and keyboard become unresponsive. Initially, I suspected that the number of environments was too high, so I switched to debug mode, but it still froze.

zhouxian, 2024-12-28:
Sorry to hear that and apologize 
Are u rendering anything? Could you provide more information for us to reproduce?

formoree, 2024-12-30:
Sorry, I tested some other code frameworks. When using the IsaacGym framework, segmentation faults frequently occurred, and there was one instance where the system froze. It should be a hardware issue. Thank you very much for your response.

formoree, 2024-12-31:
I tried again with a computer equipped with an RTX 4090 graphics card. After installing Genesis following the tutorial and attempting to run the example code go2_train.py from the locomotion module, I encountered a "Floating point exception (core dumped)" error. Could you help me resolve this issue?

python version: 3.9
Cuda version: 12.4
Driver version:550.142