Genesis-Embodied-AI/genesis-world #373: [Feature] Blood, vessels and heart simulations

lhemerly, 2024-12-28:
I want to expand Genesis for realistic simulations of vessels, blood and heart.

For this we would need, at least:
1 - Realistic blood simulation - Custom liquid materials solving with Power-Law, Casson, Carreau-Yasuda, etc
2 - Pulsatile flow - Custom emitter solving with Womersley, Navier-Stokes, FSI, etc

Of course, for a realistic simulation we may need more features like advanced Couplers but those two initial features would be a great starting point.

If you want to point the way you usually do this I can try to create the PR.

zhouxian, 2024-12-28:
This sounds pretty cool!

A few questions:
- we develop the whole framework in taichi so enjoy GPU acceleration -- the syntax is identical to python but it's still certainly work to understand everything about taichi. Is this a problem to you?
- Do you have existing implementations that you want to translate and merge into Genesis or this is something you will have to implement completely from scratch?

lhemerly, 2024-12-28:
I can give Taichi a try! 
I don't have existing implementations but those solvers are well known I think it will be more of how to adapt to the Genesis engine than anything else.

For example, let's take the Womersley pulsatile flow. 
It descibres a pulsatile flow in cylindrical vessels with a radial velocity profile and time-dependant oscilations. My idea is:
- Inherit the baseline Emitter class.
- Breakdown the shape into N(according to the desired accuracy) radial Emitters and precompute the base velocity for each radial using Womersley. This allows parallelism.
- The Womersley class applies a sinusoidal (the same for the entire shape) to the base velocity into each radial emitter for each dt.

There's other approaches though. We could, for example, have one Emitter and after it calls the emit method we could iterate over all particles and adjust their velocities.

zhouxian, 2024-12-28:
It would be good to add another solver class to handle these , but I’ll leave you to decide it :)

lhemerly, 2024-12-28:
I think the SPH-Solver plus Couplers should be enough but will give it some thought. Still navigating the codebase!