Genesis-Embodied-AI/genesis-world #374: MuJoCo actuators/motors

cbullo, 2024-12-28:
Hi,
I have a robot which uses MuJoCo actuators together with fixed tendons to implement constraints between joints. 
Genesis seems to just ignore these and only parses actual joints.
Are they not supported at all? Is there any workaround?

Here's my robot definition:
```
<mujoco>
    <compiler autolimits="true" />

    <asset>
        <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1=".1 .2 .3"
            rgb2=".2 .3 .4" />
        <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2" />
    </asset>

    <default>
        <joint damping=".03" actuatorfrcrange="-0.5 0.5" />
        <default class="wheel">
            <geom type="cylinder" size=".033 .014" rgba=".5 .5 1 1" />
        </default>
        <default class="rod">
            <geom type="box" size=".01 .01 .17" rgba=".5 .5 1 1" density="100"/>
        </default>
    </default>

    <worldbody>
        <body name="robot" pos="0 0 .203" >
            <freejoint />
            <geom type="box" size=".11 .08 .07" mass=".1"/>
            <body name="left_rod" pos="0 -.04 0">
                <joint name="left_d" type="slide" range = "-0.065 0.065"/>
                <geom class="rod" />

                <body name="left wheel" pos="0 -0.07 -0.17" zaxis="0 1 0">
                    <joint name="left_r" />
                    <geom class="wheel" />
                </body>
            </body>
            <body name="right_rod" pos="0 .04 0">
                <joint name="right_d" type="slide" range = "-0.065 0.065"/>
                <geom class="rod" />

                <body name="right wheel" pos="0 0.07 -0.17" zaxis="0 1 0">
                    <joint name="right_r" />
                    <geom class="wheel" />
                </body>
            </body>
        </body>
    </worldbody>

    <tendon>
        <fixed name="left_m1">
            <joint joint="left_r" coef="1" />
            <joint joint="left_d" coef="1" />
        </fixed>
        <fixed name="left_m2">
            <joint joint="left_r" coef="1" />
            <joint joint="left_d" coef="-1" />
        </fixed>
        <fixed name="right_m1">
            <joint joint="right_r" coef="1" />
            <joint joint="right_d" coef="1" />
        </fixed>
        <fixed name="right_m2">
            <joint joint="right_r" coef="1" />
            <joint joint="right_d" coef="-1" />
        </fixed>
    </tendon>

    <actuator>
        <motor name="left_m1_motor" tendon="left_m1" ctrlrange="-100 100"/>
        <motor name="left_m2_motor" tendon="left_m2"  ctrlrange="-100 100"/>
        <motor name="right_m1_motor" tendon="right_m1" ctrlrange="-100 100"/>
        <motor name="right_m2_motor" tendon="right_m2" ctrlrange="-100 100"/>
    </actuator>

</mujoco>
```

zhouxian, 2024-12-28:
Hi! thanks for asking :) tendon is not supported yet. We will add this soon. Should not be difficult implementation wise, but we are a little bit short of manpower in the coming weeks :(

cbullo, 2024-12-28:
How about actuators/motors?

zhouxian, 2024-12-28:
there's not concept of actuators/motors in genesis. Unlike mujoco where you need to pre-define these, _every_ joints in genesis are controllable in a very intuitive way: position control / velocity control / torque control
see: https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/control_your_robot.html

cbullo, 2024-12-28:
Oh, ok. So if I understand correctly, I would control left_m*, right_m* _tendons_ (if they were supported) instead of actuators, in case of my robot.

Hippozhibos, 2025-01-04:
Hi developers,

I’ve been following the discussions about muscle and tendon modeling (e.g., #157 and #287), and I noticed many users are interested in these features, but they’re currently not supported in Genesis.

Given that this seems like a valuable extension for the framework, I was wondering if it would be possible for the community to contribute towards implementing these features. Specifically:
1. Do you think it's feasible to add muscle and tendon simulation support to Genesis?
2. Are there any suggestions on how we could start building these features, possibly within the existing codebase?

I’d love to hear your thoughts, and if anyone from the community is interested in joining the effort, that would be great!

Thanks for your time and guidance.

duburcqa, 2025-11-20:
@Milotrince that would be awesome