Genesis-Embodied-AI/genesis-world #375: How to load camera from mjcf?

parzival2108, 2024-12-28:
Does the genesis library have  function similar to this: scene.add_camera_from_mjcf("drone_camera")?

I added camera to the xml file; but it wont get loaded? My fault, or way to attach a camera onto robot frame here in genesis?

added this to xml file:
      <camera name="drone_camera" pos="0.02 0 0" xyaxes="1 0 0 0 1 0" fovy="90"/>

zhouxian, 2024-12-28:
We don't support parsing camera yet. You'll have to create it manually :(

parzival2108, 2024-12-28:
> We don't support parsing camera yet. You'll have to create it manually :(

I can add it myself manually; how can I contribute to this repo; I can add the setup for camera parsing, but this is my first time doing this with github. So I am not sure what is the process.

zhouxian, 2024-12-28:
I am not sure if supporting Mujoco camera is something we want to do right now, as logically it’s a bit incompatible with the way we design cameras…

I would suggest you implement a parser and directly call genesis ‘add_camera’ from the parser result

ziyanx02, 2024-12-29:
You can follow this [guide](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/CONTRIBUTING.md) if you would like to contribute. However, please note that not every proposed feature will be merged into the main branch. Thank you for your interest in contributing to Genesis!

closed 2024-12-29 by ziyanx02