Genesis-Embodied-AI/genesis-world #376: Created a town with small boxes, requires separation of entities within the environment!

parzival2108, 2024-12-28:
Does the genesis system does separation of entities on the environment? 
I created a small town with small boxes, but problem is that when I place 20 dogs infront of the house, and do parallel simulation. 
It becomes extremely heavy! I believe genesis probably copy paste 20 different towns too on top of each other; which makes the houses looks much thicker!!!! 

There must be separation between parallelizing robot dogs vs the other entities such as sphere or boxes. 

https://github.com/user-attachments/assets/5b61000f-c29f-4d18-9736-0bcb3348c598

zhouxian, 2024-12-28:
try spacing=(0,0) when you build env?

parzival2108, 2024-12-28:
> try spacing=(0,0) when you build env?

Tried it! Works perfectly fine!!! Thank you

closed 2024-12-28 by zhouxian